zephyrproject-rtos/zephyr #23897: Typo in linker.ld for NXP i.MX RT

arvid-r, 2020-03-29:
**Describe the bug**
There is a typo in `linker.ld` for NXP i.MX RT platform since the conversion to new Devicetree macros. The start adress macro is used instead of the size macro for the ITCM and DTCM sections, which renders very large sections for these memories.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/hello_world
2. mkdir build; cd build
3. west build  -b mimxrt_1020evk
4. See output
```
[2/6] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           OCRAM:          0 GB       128 KB      0.00%
            DTCM:          0 GB       512 MB      0.00%
            ITCM:          0 GB         0 GB       nan%
           FLASH:       20964 B         8 MB      0.25%
            SRAM:        3916 B        32 MB      0.01%
        IDT_LIST:          72 B         2 KB      3.52%
```

**Expected behavior**
```
west build -b mimxrt1020_evk ..
[2/6] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           OCRAM:          0 GB       128 KB      0.00%
            DTCM:          0 GB        64 KB      0.00%
            ITCM:          0 GB        64 KB      0.00%
           FLASH:       20964 B         8 MB      0.25%
            SRAM:        3916 B        32 MB      0.01%
        IDT_LIST:          72 B         2 KB      3.52%
[6/6] Linking C executable zephyr/zephyr.elf
```

**Environment (please complete the following information):**
MacOS and gnu toolchain, but should be the same for all OSs I guess

Easy fix. I will prepare a PR if no-one else does.

galak, 2020-03-29:
> Easy fix. I will prepare a PR if no-one else does.

A PR would be appreciated.

arvid-r, 2020-03-30:
#23902

closed 2020-03-30 by galak