zephyrproject-rtos/zephyr #23907: Shell overdo argument parsing in some cases

nordic-krch, 2020-03-30:
**Describe the scenario**
If shell is used to implement AT commands, some commands are parsed incorrectly. It is because shell argument parser attempts to extract all parameters into separate strings and support quotation characters so it parses string like `command arg1 "arg2 with spaces"` into 3 arguments with last one containing spaces. Quotations must be escaped if they are just quotations (e.g. `command arg1\"ABC\"`). For AT case it would actually be more convenient if shell passes parameters as single string to the command handler.

Example (assuming that `at` is a registered command with handler):
command: `at AT+ABC="some string","FOO", "foo"`

Now is parsed as:
`argv[0]=at`
`argv[1]=AT+ABC=some string,FOO,`
`argv[2]=foo`

Expected:
`argv[0]=at`
`argv[1]=AT+ABC="some string","FOO", "foo"`

**Proposed solution**
Add flags to command registration and flag that will tell the parser to treat all characters following n'th argument as the last argument.
Add flag `SHELL_CMD_MERGE_ARGS(offset)`. Offset written on 4 bits should be more than enough.
Extend registration command to allow passing `flags`.

nordic-krch, 2020-03-30:
fyi @stig-bjorlykke

nordic-krch, 2020-03-30:
@jakub-uC  @Vudentz what do you think about it? feedback welcomed. Other solution proposals or maybe different perspective to the problem.

jakub-uC, 2020-03-30:
@nordic-krch : The easiest way, in my opinion, would be to implement a command like `select` which you can use to dynamically modify parameter set by Kconfig's `SHELL_ARGC_MAX`.
And similar to `select` you could use some escape code to bring back default value.

PR #24329 (2020-05-08): shell: Refactor command execution to enable raw arguments
 Added special flag that can be used to indicate that optional arguments are passed without any parsing (e.g. quotation marks removal). Modified execute command to parse command line buffer argument by argument.

After this change it is possible to forward whole command to command handler (using select).

It is now possible to specify command like:

`SHELL_CMD_ARG_REGISTER(python, NULL, "Python", cmd_python, 1,  SHELL_OPT_ARG_RAW);`

Then call
```
select python
```

and whole content from the prompt will be passed directly to the handle [... cut]
closed 2020-05-08 by galak