zephyrproject-rtos/zephyr #23917: Kconfig: problems with using backslash-escapes in "default" value of "string" option

pfalcon, 2020-03-30:
I'm trying to use following snippet in a Kconfig:

~~~
config SAMPLE_BIG_HTTP_DL_HASH
	string "Hash of downloaded file"
	default "\x33\x7c\x37\xd7\xec\x00\x34\x84\x14\x22\x4b\xaa\x6b\xdb\x2d\x43\xf2\xa3\x4e\xf5\x67\x6b\xaf\xcd\xca\xd9\x16\xf1\x48\xb5\xb3\x17"
~~~

However, zephyr/.config with that setting gets:

~~~
CONFIG_SAMPLE_BIG_HTTP_DL_HASH="x33x7cx37xd7xecx00x34x84x14x22x4bxaax6bxdbx2dx43xf2xa3x4exf5x67x6bxafxcdxcaxd9x16xf1x48xb5xb3x17"
~~~

When I try to used double-backslashes in Kconfig:
~~~
	default "\\x33\\x7c\\x37\\xd7\\xec\\x00\\x34\\x84\\x14\\x22\\x4b\\xaa\\x6b\\xdb\\x2d\\x43\\xf2\\xa3\\x4e\\xf5\\x67\\x6b\\xaf\\xcd\\xca\\xd9\\x16\\xf1\\x48\\xb5\\xb3\\x17"
~~~

I get in zephyr/.config:
~~~
CONFIG_SAMPLE_BIG_HTTP_DL_HASH="\\x33\\x7c\\x37\\xd7\\xec\\x00\\x34\\x84\\x14\\x22\\x4b\\xaa\\x6b\\xdb\\x2d\\x43\\xf2\\xa3\\x4e\\xf5\\x67\\x6b\\xaf\\xcd\\xca\\xd9\\x16\\xf1\\x48\\xb5\\xb3\\x17"
~~~

Complete reproducible case can be found in this branch: https://github.com/pfalcon/zephyr/commits/big_http_download-kconfigurable

pfalcon, 2020-03-30:
@ulfalizer, Hopefully you can look into this.

pfalcon, 2020-04-02:
We discussed this issue yesterday on Slack with @ulfalizer. Just to summarize it, the current situation is that you can't pass a string with a *single* backslash from Kconfig default value down to `autoconf.h` file generated by kconfiglib (indeed, `autoconf.h` is the real problem spot, as that's what gets included from C source, not `.config`). Currently, you can pass either 0 or 2 slashes, with 1 being gapped.

Based on @ulfalizer guidance thru the kconfiglib source, I came with the following patch which satisfies my concern:

~~~
--- a/scripts/kconfig/kconfiglib.py
+++ b/scripts/kconfig/kconfiglib.py
@@ -1488,7 +1488,7 @@ class Kconfig(object):
 
             elif sym.orig_type is STRING:
                 add('#define {}{} "{}"\n'
-                    .format(self.config_prefix, sym.name, escape(val)))
+                    .format(self.config_prefix, sym.name, val))
 
             else:  # sym.orig_type in _INT_HEX:
                 if sym.orig_type is HEX and \
~~~

With this change, things on the Kconfig side need to be extra-quoted, i.e. `default "\\x33\\x7c\\x37"` leads to `#define CONFIG_SAMPLE_BIG_HTTP_DL_HASH "\x33\x7c\x37"`. This extra level of quoting is completely ok from my side, as long as the rules for quoting are consistent and monotonic (there's no random gaps between 0 and 2 number of slashes output).

One issue is that the change as presented above applies to the `autoconf.h` output and not `.config` output. This is again based on @ulfalizer's concern that .config is normally included in Makefile's, and they may have other quoting requirements than C. Another obvious concern is that this change may lead to a different behavior than the referece Kconfig implementation written in C (as used by Linux kernel, etc.).

My reply to the latter concern would be that it could be the case that processing backslash quotes may be not much used in the original kernel Kconfig, and thus not very well defined. It could be the case that being "bug-to-bug compatible" with the original Kconfig isn't really required and there can be more useful behavior proposed here.

I could also understand the position of being "bug-to-bug compatible", as after all, that simplifies testing, by allowing to just compare the outputs of 2 systems for exact matches under any input thrown at them. In this case, I would propose to add another value type with the desired behavior, e.g.:

~~~
config SAMPLE_BIG_HTTP_DL_HASH
	rawstring "Hash of downloaded file"
	default "\\x33\\x7c\\x37..."
~~~

Note the **rawstring** type. This would be an extension not used by Linux kernel, but which is useful for Zephyr and other project with deeper C integration needs (like, need to inject raw/binary C strings in the config, without relying that there will be any "frontend" on the application side which can accept human-readable representation as input and parse it internally).

All in all, recording these findings and ideas, and handing over to @ulfalizer for further analysis and decision-making. Thanks.

pfalcon, 2020-04-06:
For reference, PR affected by this issue: https://github.com/zephyrproject-rtos/zephyr/pull/24124#issuecomment-609791549 , where I had to scale down originally configurability effort, and keep some hardcoded things as such.

carlescufi, 2020-04-07:
In the meantime @pfalcon, does `0x<long number>` work?

pfalcon, 2020-04-07:
@carlescufi 

> In the meantime @pfalcon, does 0x<long number> work?

No, the talk here is about being able to specify arbitrary-length binary string, not limited-length binary number. (If you hint at being able to break down a long binary string into a number of integer-values options, like SAMPLE_BIG_HTTP_DL_HASH_INT_0, SAMPLE_BIG_HTTP_DL_HASH_INT_1, ... SAMPLE_BIG_HTTP_DL_HASH_INT_100, then apparently it's a pretty unbeautiful and not scalable workaround.)

pfalcon, 2020-07-01:
@ulfalizer: Ping. It occurred to me that while we discussed this issue at length on the Slack, this ticket didn't receive any comment from you.

closed 2020-10-03 by github-actions[bot]