zephyrproject-rtos/zephyr #23919: sanitycheck samples/drivers/entropy/sample.drivers.entropy fails

chrta, 2020-03-30:
**Describe the bug**

The sanitycheck with the test samples/drivers/entropy/sample.drivers.entropy always fails for the board efm32pg_stk3402a. This seems to also be the case for other boards.

It seems that the regular expression `"(  0x([0-9a-f]){2}){9}"` in the file `sample.yaml` expects two spaces at the beginning of the line, but there are no spaces, according to the file `handler.log`.
According to `main.c`, the two spaces at the beginning of the line are printed. The two spaces at the beginning of the line are present in the console when using minicom.

**To Reproduce**
Steps to reproduce the behavior:
1.  `./scripts/sanitycheck -p efm32pg_stk3402a --device-testing --device-serial /dev/ttyACM1 --west-flash="--reset-after-load" --test samples/drivers/entropy/sample.drivers.entropy -v`

**Expected behavior**

The test should be PASSED

**Impact**

Only low impact, since only the sanitycheck reports that the entropy driver is not working, even if it is.

**Console output**

Console output:
```
...
INFO    - 1/1 efm32pg_stk3402a          samples/drivers/entropy/sample.drivers.entropy     FAILED Timeout (device 60.825s)
...
```
samples/drivers/entropy/sample.drivers.entropy/handler.log:
```
0x8e  0x04  0xac  0x99  0x63  0xa4  0xcd  0x0a  0xa0
*** Booting Zephyr OS build zephyr-v2.2.0-894-gba6dc5bc2572  ***
Entropy Example! arm
entropy device is 0x20000f50, name is TRNG0
0xe1  0xc3  0xa9  0x10  0x16  0xe1  0xf1  0x41  0xfa
0x1d  0x71  0xfc  0x4f  0xc0  0x9d  0x31  0x57  0xb2
0xeb  0x83  0xb8  0xe1  0x1d  0xc5  0x22  0x14  0x83
0x24  0x71  0x97  0x22  0xc5  0x9b  0x66  0xa1  0xb4
0x03  0x17  0xa0  0x3b  0xfb  0xa9  0xc4  0xbd  0x6d
0x2e  0xc6  0xe0  0x09  0x26  0xc2  0x42  0x24  0xd3
...
```

**Environment:**
 - Ubuntu 18.04
 - Toolchain Zephyr SDK
 - Commit SHA ba6dc5bc25720715b0e8957cca19e12778ac05a7 is current master

nashif, 2020-03-31:
Does this solve the problem?

```diff
--- a/samples/drivers/entropy/sample.yaml
+++ b/samples/drivers/entropy/sample.yaml
@@ -11,4 +11,4 @@ tests:
       regex:
         - "Entropy Example! (.*)"
         - "entropy device is (.*), name is (.*)"
-        - "(  0x([0-9a-f]){2}){9}"
+        - "(\s*0x([0-9a-f]){2}){9}"
```

chrta, 2020-03-31:
This produces:
```
/home/user/zephyrproject/zephyr/samples/drivers/entropy/sample.yaml:13:13: error: found unknown escape character 's' (note None context @/home/user/zephyrproject/zephyr/samples/drivers/entropy/sample.yaml:13:10 while scanning a double-quoted scalar)
ERROR   - /home/user/zephyrproject/zephyr/samples/drivers/entropy/sample.yaml: can't load (skipping): while scanning a double-quoted scalar
  in "/home/user/zephyrproject/zephyr/samples/drivers/entropy/sample.yaml", line 14, column 11
found unknown escape character 's'
  in "/home/user/zephyrproject/zephyr/samples/drivers/entropy/sample.yaml", line 14, column 14
```
There seems to be escaping missing. The following works and fixes the problem:
```diff
--- a/samples/drivers/entropy/sample.yaml
+++ b/samples/drivers/entropy/sample.yaml
@@ -11,4 +11,4 @@ tests:
       regex:
         - "Entropy Example! (.*)"
         - "entropy device is (.*), name is (.*)"
-        - "(  0x([0-9a-f]){2}){9}"
+        - "(\\s*0x([0-9a-f]){2}){9}"
```

Result:
```
INFO    - Adding tasks to the queue...
INFO    - 1/1 efm32pg_stk3402a          samples/drivers/entropy/sample.drivers.entropy     PASSED (device 0.782s)

INFO    - 1 of 1 tests passed (100.00%), 0 failed, 0 skipped with 0 warnings in 5.62 seconds
INFO    - In total 1 test cases were executed on 1 out of total 241 platforms (0.41%)

```

nashif, 2020-04-01:
Can you submit a PR? :)

PR #24006 (2020-04-03): samples: drivers/entropy yaml regex fix
The last regex in the yaml file did not match when executing the test on
an embedded target. The expected two spaces are not present at the
beginning of the line.

Fixes #23919
commit 1bf5d246 (2020-04-03): samples: drivers/entropy yaml regex fix

The last regex in the yaml file did not match when executing the test on
an embedded target. The expected two spaces are not present at the
beginning of the line.

Fixes #23919
closed 2020-04-03 by nashif