zephyrproject-rtos/zephyr #23928: Flash device FLASH_CTRL not found

mtahirbutt, 2020-03-30:
**Describe the bug**
[00:00:00.005,000] <err> mcuboot: Flash device FLASH_CTRL not found error. I have want to build that to further proceed towards updatehub app. But I am stopped here. Please someone help in this regard.

What have you tried to diagnose or workaround this issue?

I tried the following method to resolve the issue.
**To Reproduce**
Steps to reproduce the behavior:

tahir@ubuntu:~/zephyrproject$ cd zephyr/
tahir@ubuntu:~/zephyrproject/zephyr$ source zephyr-env.sh
tahir@ubuntu:~/zephyrproject/zephyr$ cd ..
tahir@ubuntu:~/zephyrproject$ cd bootloader/
tahir@ubuntu:~/zephyrproject/bootloader$ cd mcuboot/
tahir@ubuntu:~/zephyrproject/bootloader/mcuboot$ cd boot/
tahir@ubuntu:~/zephyrproject/bootloader/mcuboot/boot$ cd zephyr/
tahir@ubuntu:~/zephyrproject/bootloader/mcuboot/boot/zephyr$ git branch

(HEAD detached at 1084100)
manifest-rev
tahir@ubuntu:~/zephyrproject/bootloader/mcuboot/boot/zephyr/build_stm32l4x$ cmake -GNinja -DBOARD=disco_l475_iot1 ..
-- Application: /home/tahir/zephyrproject/bootloader/mcuboot/boot/zephyr
-- Zephyr version: 2.2.99
-- Found PythonInterp: /usr/bin/python3.6 (found suitable version "3.6.9", minimum required is "3.6")
-- Board: disco_l475_iot1
-- Found west: /home/tahir/.local/bin/west (found suitable version "0.7.2", minimum required is "0.6.0")
-- Found toolchain: zephyr (/home/tahir/zephyr-sdk-0.11.2)
-- Loading /home/tahir/zephyrproject/zephyr/boards/arm/disco_l475_iot1/disco_l475_iot1.dts as base
-- Overlaying /home/tahir/zephyrproject/bootloader/mcuboot/boot/zephyr/dts.overlay
Devicetree header saved to '/home/tahir/zephyrproject/bootloader/mcuboot/boot/zephyr/build_stm32l4x/zephyr/include/generated/devicetree_unfixed.h'
Parsing /home/tahir/zephyrproject/bootloader/mcuboot/boot/zephyr/Kconfig
Loaded configuration '/home/tahir/zephyrproject/zephyr/boards/arm/disco_l475_iot1/disco_l475_iot1_defconfig'
Merged configuration '/home/tahir/zephyrproject/bootloader/mcuboot/boot/zephyr/prj.conf'
Configuration saved to '/home/tahir/zephyrproject/bootloader/mcuboot/boot/zephyr/build_stm32l4x/zephyr/.config'
Kconfig header saved to '/home/tahir/zephyrproject/bootloader/mcuboot/boot/zephyr/build_stm32l4x/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/tahir/zephyr-sdk-0.11.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Cache files will be written to: /home/tahir/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tahir/zephyrproject/bootloader/mcuboot/boot/zephyr/build_stm32l4x
tahir@ubuntu:~/zephyrproject/bootloader/mcuboot/boot/zephyr/build_stm32l4x$ ninja
[1/235] Preparing syscall dependency handling
[230/235] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region Used Size Region Size %age Used
FLASH: 46896 B 64 KB 71.56%
SRAM: 29596 B 96 KB 30.11%
IDT_LIST: 168 B 2 KB 8.20%
[235/235] Linking C executable zephyr/zephyr.elf
tahir@ubuntu:~/zephyrproject/bootloader/mcuboot/boot/zephyr/build_stm32l4x$ cd zephyr/
tahir@ubuntu:~/zephyrproject/bootloader/mcuboot/boot/zephyr/build_stm32l4x/zephyr$ grep -nr . -e FLASH_CTRL
Binary file ./CMakeFiles/zephyr.dir/subsys/storage/flash_map/flash_map_default.c.obj matches
Binary file ./zephyr.elf matches
Binary file ./zephyr_prebuilt.elf matches
Binary file ./libzephyr.a matches
Binary file ./zephyr.bin matches
./disco_l475_iot1.dts.pre.tmp:44: label = "FLASH_CTRL";
./zephyr.dts:42: label = "FLASH_CTRL";
./include/generated/devicetree_unfixed.h:1470:#define DT_ST_STM32L4_FLASH_CONTROLLER_40022000_LABEL "FLASH_CTRL"
./include/generated/devicetree_unfixed.h:1552:#define DT_FLASH_AREA_MCUBOOT_DEV "FLASH_CTRL"
./include/generated/devicetree_unfixed.h:1559:#define DT_FLASH_AREA_0_DEV "FLASH_CTRL"
./include/generated/devicetree_unfixed.h:1588:#define DT_FLASH_AREA_IMAGE_0_DEV "FLASH_CTRL"
./include/generated/devicetree_unfixed.h:1595:#define DT_FLASH_AREA_1_DEV "FLASH_CTRL"
./include/generated/devicetree_unfixed.h:1624:#define DT_FLASH_AREA_IMAGE_1_DEV "FLASH_CTRL"
./include/generated/devicetree_unfixed.h:1631:#define DT_FLASH_AREA_2_DEV "FLASH_CTRL"
./include/generated/devicetree_unfixed.h:1660:#define DT_FL

tahir@ubuntu:~/zephyrproject/bootloader/mcuboot/boot/zephyr/build_stm32l4x/zephyr$ sha256sum zephyr.bin zephyr.elf
f9a0ac6bda0d9a528a58ec0561a04b58180767a7e79483220fa43500044e0eed zephyr.bin
bc2c3033420c52d91e0c9a2d59d6bc6f6c80d431e8f1d316c7cae39a030e972f zephyr.elf
tahir@ubuntu:~/zephyrproject/bootloader/mcuboot/boot/zephyr/build_stm32l4x/zephyr$ west flash
FATAL ERROR: --build-dir was not given, and neither /home/tahir/zephyrproject/bootloader/mcuboot/boot/zephyr/build_stm32l4x/zephyr nor /home/tahir/zephyrproject/bootloader/mcuboot/boot/zephyr/build_stm32l4x/zephyr/build are zephyr build directories.
tahir@ubuntu:~/zephyrproject/bootloader/mcuboot/boot/zephyr/build_stm32l4x/zephyr$ cd ..
tahir@ubuntu:~/zephyrproject/bootloader/mcuboot/boot/zephyr/build_stm32l4x$ west flash

**Expected behavior**
Bootlooder successfully loads into the application and no error message is seen on the console.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

I am stopped in proceeding towards testing updatehub app.

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

[00:00:00.005,000] <inf> mcuboot: Starting bootloader
[00:00:00.005,000] <err> mcuboot: Flash device FLASH_CTRL not found

**Environment (please complete the following information):**
 - board- disco_l475_iot1
 - OS: (Ubuntu 18.04.4 LTS)
 - Toolchain (e.g Zephyr SDK, 0.11-2)
 - Commit SHA: commit 1c79096559f5b95f7153cd6784459734c8bb76b9 (HEAD -> master, origin/master)

I need to resolve that so that updatehub app is able to recognize the flash partitions.
Add any other context about the problem here.

erwango, 2020-03-30:
Could you check if this is fixed by https://github.com/zephyrproject-rtos/zephyr/pull/23679 ?

mtahirbutt, 2020-03-31:
issue is resolved thanks. by replacing with DT_FLASH_DEV_NAME inside zephyr/drivers/flash_stm32.c

DEVICE_AND_API_INIT(stm32_flash, DT_FLASH_DEV_NAME,
...........................

closed 2020-03-31 by mtahirbutt