zephyrproject-rtos/zephyr #23930: Question: Cortex-M7 revision r0p1 errata

pftbest, 2020-03-30:
Some Cortex-M7 chips has errata 837070 "Increasing priority using a write to BASEPRI does not take effect immediately".

Is there any mitigation for this in zephyr kernel? I can see a lot of writes to BASEPRI in the source code but it's not clear to me if they are affected or not.

Other projects implement a workaround for this, for example FreeRTOS has a separate port for r0p1.

stephanosio, 2020-03-31:
cc @ioannisg

ioannisg, 2020-03-31:
Thanks for posting this, @pftbest .

As far as I can see from this errata:

> This effect is only limited to the next instruction. Subsequent instructions are guaranteed to see the new boosted priority.

So, it seems to me that even though we do not _directly_ account for this errata in the code, MSR to BASEPRI is followed by an ISB instruction; (this was done for extra protection so even fetching instructions or reading registers / ALU operations are not done before MSR BASEPRI is completed) the latter takes care of this errata as well. 

@stephanosio ^^

stephanosio, 2020-03-31:
It seems we do indeed have `isb` after all boosts to `BASEPRI`, as far as the arch implementation goes (e.g. in `arch_irq_lock`), and given that this errata effectively delays interrupt locking until the next instruction is executed, where that next instruction is always `isb`, this should not be a problem.

https://github.com/zephyrproject-rtos/zephyr/blob/33928f18ae063fa8cb97ebd35481d7a7db04e927/include/arch/arm/aarch32/asm_inline_gcc.h#L59-L60

stephanosio, 2020-03-31:
@ioannisg What is rather curious is that FreeRTOS Cortex-M7 r0p1 port (the one with errata workaround) implements the `cpsid i` and `cpsie e` workaround despite having `dsb` and `isb` after `msr basepri, ...`.

https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/88e32327e975ddde97c390bc5b6c1f8e7d9d239e/portable/IAR/ARM_CM7/r0p1/portmacro.h#L131-L136

ioannisg, 2020-04-21:
> @ioannisg What is rather curious is that FreeRTOS Cortex-M7 r0p1 port (the one with errata workaround) implements the `cpsid i` and `cpsie e` workaround despite having `dsb` and `isb` after `msr [...]

Might just be done for extra protection. Hard to assess with un-commented code.

ioannisg, 2020-04-21:
@pftbest I guess you can close this question if you feel the response was adequate

pftbest, 2020-04-21:
Ok, thank you.

closed 2020-04-21 by pftbest