zephyrproject-rtos/zephyr #23953: Question: How is pdata.tsize initialized in zephyr/subsys/usb/usb_transfer.c?

tluio, 2020-03-31:
In function usb_transfer_sync, pdata.tsize is returned. I wonder how it is initialized. Is it initialized through DMA?

jfischer-phytec-iot, 2020-04-01:
For `usb_transfer_sync()`, which blocks until finished, pointer to the pdata is `cb_data` and `pdata.tsize` will be assigned in the callback of [usb_transfer()](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/usb_transfer.c#L296).

tluio, 2020-04-01:
Thank you very much! Would you please tell me how usb_transfer_sync() blocks.

emob-nordic, 2020-04-01:
There is semaphore which is [taken](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/usb_transfer.c#L317) and later [released](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/usb_transfer.c#L301).

tluio, 2020-04-01:
@emob-nordic Thank you! Now I think I should close this issue.

closed 2020-04-01 by tluio