zephyrproject-rtos/zephyr #23980: Nordic USB driver: last fragment sometimes dropped for OUT control endpoint

mkarhumaa, 2020-04-01:
**Describe the bug**
Nordic USB driver seems to be sometimes dropping last fragment of fragmented transfer to OUT control endpoint. This appeared when I was testing hci_usb sample and implementing vendor specific HCI command with parameters length bigger than 64 octets (meaning it gets fragmented). I ran into problem that sometimes I did not get HCI Command Complete event which caused HCI TX timeout to trigger in our Bluetooth stack. Further debugging revealed that actually Zephyr's USB stack did not always get the last fragment of OUT transfer and `residue` does not get to 0 and does not get passed to handler.

I am running hci_usb sample on nRF52840 DK. This seems to be reproducible with `master` and `v2.2-branch`.

Following patch adds more logging:
```
From ae66f49133a1bd8e0ceedff54755848803249b04 Mon Sep 17 00:00:00 2001
From: Matias Karhumaa <matias.karhumaa@gmail.com>
Date: Wed, 1 Apr 2020 06:13:57 +0300
Subject: [PATCH] usb: Add logging to nordic driver

---
 samples/bluetooth/hci_usb/prj.conf | 8 ++++++++
 subsys/usb/usb_device.c            | 6 +++++-
 2 files changed, 13 insertions(+), 1 deletion(-)

diff --git a/samples/bluetooth/hci_usb/prj.conf b/samples/bluetooth/hci_usb/prj.conf
index 1c19e32540..b54b662861 100644
--- a/samples/bluetooth/hci_usb/prj.conf
+++ b/samples/bluetooth/hci_usb/prj.conf
@@ -9,3 +9,11 @@ CONFIG_BT_HCI_RAW=y
 CONFIG_USB=y
 CONFIG_USB_DEVICE_STACK=y
 CONFIG_USB_DEVICE_BLUETOOTH=y
+
+CONFIG_BT_DEBUG_LOG=y
+CONFIG_USB_DEVICE_LOG_LEVEL_DBG=y
+CONFIG_USB_DRIVER_LOG_LEVEL_DBG=y
+CONFIG_LOG_PROCESS_THREAD_STACK_SIZE=2048
+CONFIG_LOG_BUFFER_SIZE=32768
+CONFIG_LOG_STRDUP_BUF_COUNT=64
+
diff --git a/subsys/usb/usb_device.c b/subsys/usb/usb_device.c
index a126116e04..b0ba772245 100644
--- a/subsys/usb/usb_device.c
+++ b/subsys/usb/usb_device.c
@@ -258,7 +258,7 @@ static void usb_handle_control_transfer(u8_t ep,
 
 	if (ep == USB_CONTROL_OUT_EP0 && ep_status == USB_DC_EP_SETUP) {
 		u16_t length;
-
+		LOG_DBG("OUT transfer, Setup packet,");
 		/*
 		 * OUT transfer, Setup packet,
 		 * reset request message state machine
@@ -306,9 +306,11 @@ static void usb_handle_control_transfer(u8_t ep,
 		/* Send first part (possibly a zero-length status message) */
 		usb_data_to_host(length);
 	} else if (ep == USB_CONTROL_OUT_EP0) {
+		LOG_DBG("OUT transfer, data or status packets");
 		/* OUT transfer, data or status packets */
 		if (usb_dev.data_buf_residue <= 0) {
 			/* absorb zero-length status message */
+			LOG_DBG("absorb zero-length status message");
 			if (usb_dc_ep_read(USB_CONTROL_OUT_EP0,
 					   usb_dev.data_buf, 0, &chunk) < 0) {
 				LOG_DBG("Read DATA Packet failed");
@@ -328,7 +330,9 @@ static void usb_handle_control_transfer(u8_t ep,
 
 		usb_dev.data_buf += chunk;
 		usb_dev.data_buf_residue -= chunk;
+		LOG_DBG("Residue: %d", usb_dev.data_buf_residue);
 		if (usb_dev.data_buf_residue == 0) {
+			LOG_DBG("Received all, send data to handler");
 			/* Received all, send data to handler */
 			usb_dev.data_buf = usb_dev.req_data;
 			if (!usb_handle_request(setup,
-- 
2.17.1
```

While I initially discovered this bug in normal use using proprietary Bluetooth stack, hcitool can be used to reproduce this easily. Following command will send vendor specific command in a loop and soon it gets stuck because it is not getting Command Complete event.

```
$ while true; do sudo hcitool -i hci1 cmd 0x3f 0x0009 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02 0x02; done
```

From logs we see that `residue` does not get to 0 and buffer will not get passed to handler when getting stuck. See tail of the console log:
```
[00:01:25.439,575] <dbg> usb_nrfx.usbd_work_process_setup: SETUP: r:0 rt:32 v:0 i:0 l:76
[00:01:25.439,575] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
[00:01:25.439,575] <dbg> usb_device.usb_handle_control_transfer: OUT transfer, Setup packet,
[00:01:25.439,605] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 8
[00:01:25.439,727] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl read request
[00:01:25.439,819] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl read done: 64
[00:01:25.439,849] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x01
[00:01:25.439,880] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl read request
[00:01:25.439,910] <dbg> usb_device.usb_handle_control_transfer: OUT transfer, data or status packets
[00:01:25.439,910] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 76
[00:01:25.439,971] <dbg> usb_device.usb_handle_control_transfer: Residue: 12
[00:01:25.440,032] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl read done: 12
[00:01:25.440,063] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x01
[00:01:25.440,093] <dbg> usb_device.usb_handle_control_transfer: OUT transfer, data or status packets
[00:01:25.440,093] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 12
[00:01:25.440,124] <dbg> usb_device.usb_handle_control_transfer: Residue: 0
[00:01:25.440,155] <dbg> usb_device.usb_handle_control_transfer: Received all, send data to handler
[00:01:25.440,155] <dbg> usb_device.class_handler: bRequest 0x0, wIndex 0x0
[00:01:25.440,185] <dbg> usb_bluetooth.bluetooth_class_handler: len 76
[00:01:25.440,216] <dbg> usb_device.usb_handle_control_transfer: >> usb_data_to_host(2)
[00:01:25.440,216] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x80, len 0
[00:01:25.440,307] <dbg> usb_transfer.usb_transfer: Transfer start, ep 0x81, data 0x2000e849, dlen 29
[00:01:25.440,368] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x81, len 29
[00:01:25.441,162] <dbg> usb_nrfx.usbd_event_transfer_data: write complete, ep 0x81
[00:01:25.441,223] <dbg> usb_nrfx.usb_dc_ep_write: ep_write: ep 0x81, len 13
[00:01:25.442,169] <dbg> usb_nrfx.usbd_event_transfer_data: write complete, ep 0x81
[00:01:25.442,230] <dbg> usb_transfer.usb_transfer_work: Transfer done, ep 0x81, status 0, size 29
[00:01:25.449,401] <dbg> usb_nrfx.usbd_work_process_setup: SETUP: r:0 rt:32 v:0 i:0 l:76
[00:01:25.449,432] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
[00:01:25.449,432] <dbg> usb_device.usb_handle_control_transfer: OUT transfer, Setup packet,
[00:01:25.449,462] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 8
[00:01:25.449,554] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl read request
[00:01:25.449,645] <dbg> usb_nrfx.usbd_event_transfer_ctrl: ctrl read done: 64
[00:01:25.449,676] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x01
[00:01:25.449,707] <dbg> usb_device.usb_handle_control_transfer: OUT transfer, data or status packets
[00:01:25.449,707] <dbg> usb_nrfx.usb_dc_ep_read: ep_read: ep 0x00, maxlen 76
[00:01:25.449,737] <dbg> usb_device.usb_handle_control_transfer: Residue: 12
```

Using Wireshark we can see that when getting stuck, URB status is set to `-EPROTO`. See screnshot.

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. apply attached patch to enable more verbose logging
2. cd samples/bluetooth/hci_usb
3. west build -b nrf52840_pca10056 && west flash
4. Then use hcitool to send vendor specific hci command with payload > 64 octets, until it gets stuck

**Expected behavior**
I would expect Zephyr to deliver all the fragments of OUT control transfers.

**Impact**
This is blocker because HCI TX timeout usually causes problem on Bluetooth stack side.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - master (a3fae2f153587c552623ad4940eb6e058aeec78f)

carlescufi, 2020-04-07:
@emob-nordic have you had a chance to take a look?

emob-nordic, 2020-04-08:
@carlescufi Not yet. I will try to have a look today/tomorrow.

emob-nordic, 2020-04-10:
I've reproduced it easily. It doesn't happen sometimes, it happens always after same amount of hci commands send from host. For me it's always 1725. Digging deeper...

mkarhumaa, 2020-04-14:
@emob-nordic Yes, I noticed the same that it happens quite consistently. I haven't been able to fully track down the correlation between the length of HCI command and amount of commands needed to get stuck. For example if using HCI command of length 255 bytes (including opcode and length fields), I need to send the command only 128 times. Then with HCI command of length 254, I need 8256 HCI commands and then it gets stuck. Btw, 129*64=8256. Also I noticed that when using HCI command length near 256, _odd_ lengths trigger this around 130 consistently. Commands with _even_ length require couple of thousands of commands.

emob-nordic, 2020-04-16:
I've turned on logs in nrfx driver and at some point the driver does not create an event for reading the last 12Byte packet. I have not yet figured out why it happens. I will try to narrow the searches..

mkarhumaa, 2020-04-21:
@emob-nordic Is there any other configuration options than `CONFIG_USB_DEVICE_LOG_LEVEL_DBG` and  `CONFIG_USB_DRIVER_LOG_LEVEL_DBG` that I could try to further debug this problem?

emob-nordic, 2020-04-21:
@mkarhumaa You can enable logs in nrfx driver, this is what I did for now.
You can do this by adding
``` diff
diff --git a/samples/subsys/usb/testusb/CMakeLists.txt b/samples/subsys/usb/testusb/CMakeLists.txt
index cbab2f034a..f4fb9bd4ce 100644
--- a/samples/subsys/usb/testusb/CMakeLists.txt
+++ b/samples/subsys/usb/testusb/CMakeLists.txt
@@ -6,3 +6,6 @@ project(testusb)
 
 FILE(GLOB app_sources src/*.c)
 target_sources(app PRIVATE ${app_sources})
+
+zephyr_compile_definitions(NRFX_USBD_CONFIG_LOG_ENABLED=1)
+zephyr_compile_definitions(NRFX_USBD_CONFIG_LOG_LEVEL=4)

```
Which will extend debug logs to ll driver.

mkarhumaa, 2020-04-21:
Thank you @emob-nordic ! With these debug options I got little bit further. 

On the left is snippet from log of successful transfer. On the right side is the case that gets stuck.
[image]

From logs we can clearly see that three lines are missing in "bad" case. For some reason `USBD_INTEN_EP0DATADONE_Pos` interrupt does not trigger and `usbd_ep_data_handler` does not get called and furthermore `usbd_event_transfer_ctrl` is not called meaning that `ep_ctx->read_pending` does not get set to `true`. Later on usbd work is not scheduled because of that.

Do you have any idea what could cause this?

emob-nordic, 2020-04-21:
That was what I came up with as well but maybe described it a bit too laconic. Tbh I do not know. It may be that HW stucks somehow. I've checked if maybe EP0 are  STALLED somehow on the way but this not happen.
I wanted to test it with Nordic SDK as USB is working there for a long time. My plan was to rework nordic sdk stack to accept custom commands via EP0 and test it there. If same happen with SDK-> potential HW issue. The problem is this will take me some time to check and I do not have time for this now. Will try to find time this week.

jfischer-phytec-iot, 2020-04-22:
> While I initially discovered this bug in normal use using proprietary Bluetooth stack, hcitool can be used to reproduce this easily. Following command will send vendor specific command in a loop and [...]

Is it reproducible with hcitool when all the logging are disabled?

mkarhumaa, 2020-04-22:
> Is it reproducible with hcitool when all the logging are disabled?
@jfischer-phytec-iot Yes. It is reproducible with default configuration of hci_usb sample.

mkarhumaa, 2020-05-05:
Hi @emob-nordic,
Did you have time to test this scenario with Nordic SDK?

emob-nordic, 2020-05-05:
Hi @mkarhumaa,
Sorry but no. I am working on usb audio pr until the code freeze on May 8th. After that day I will have more time to work with bugs. Sorry for inconvenience.

mkarhumaa, 2020-05-05:
@emob-nordic Ok, thanks for update!

emob-nordic, 2020-05-19:
Ok. I know why it clogs. I do not know how to fix that but at least I know the issue. The problem is [here](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/usb/device/usb_dc_nrfx.c#L1016). After some time this condition is not true (buffer overflows or whatsoever - I did not check that yet). Because of that [nrfx_usbd_setup_data_clear](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/usb/device/usb_dc_nrfx.c#L1018) is not called. What this function does is (drums) triggers `NRF_USBD_TASK_EP0RCVOUT`. If it doesn't -> `ev_setup_data_handler` would never be called and would not trigger read of the 12 last Bytes of the custom hci command.

emob-nordic, 2020-05-20:
Hi @mkarhumaa,

The issue was introduced when addressing another issue #11232 with #11713 pr.
If You want to proceed with Your solution You can apply this temporary workaround:

``` diff
diff --git a/drivers/usb/device/usb_dc_nrfx.c b/drivers/usb/device/usb_dc_nrfx.c
index cb1fefee18..808775a8c7 100644
--- a/drivers/usb/device/usb_dc_nrfx.c
+++ b/drivers/usb/device/usb_dc_nrfx.c
@@ -1017,6 +1017,7 @@ static void usbd_event_transfer_ctrl(nrfx_usbd_evt_t const *const p_event)
 				ctx->ctrl_read_len -= ep_ctx->buf.len;
 				nrfx_usbd_setup_data_clear();
 			} else {
+				nrfx_usbd_setup_data_clear();
 				ctx->ctrl_read_len = 0U;
 			}
```

I am however working on more elegant solution as the code that was introduced in #11713 isn't quite what we want.

mkarhumaa, 2020-05-20:
Hi @emob-nordic,

Thank you, your proposed workaround seems to fix the problem!

PR #25484 (2020-05-22): usb: driver: Fix Nordic driver for fragmented control OUT transfers.
Nordic driver must allow read from control OUT endpoint by
itself. For data stage transactions with length > MPS (64Bytes)
this must be performed for each 64B + Residue data packet.

Residue - data packet with len < 64B.

The exact length of data transfer is known from wLength field
form setup packet in setup stage. Until now driver was incorrectly
initializing the length of the data stage and at some point will
not allow for next data stage.

Fixes #23980
closed 2020-05-22 by carlescufi