zephyrproject-rtos/zephyr #23989: Switching among different PHY Modes

BOZHENG001, 2020-04-01:
**Is your enhancement proposal related to a problem? Please describe.**
I am working with nRF52840 DK using Zephyr 2.0, and now i would like to see the difference among different PHY Modes. In BLE stack, there are totally 4 PHY Modes, which are LE 1M PHY, LE 2M PHY, Coded PHY (S2) and Coded PHY (S8). 

I saw some configuration choices like `CONFIG_BT_CTLR_PHY_2M` and `CONFIG_BT_CTLR_PHY_CODED`. I suppose if I would like to enable LE 2M PHY, I could easily use `CONFIG_BT_CTLR_PHY_2M=y`, and set `CONFIG_BT_CTLR_PHY_2M=n` to use LE 1M PHY. But how could I know which Coded PHY am I using, if I only enable Coded PHY by `CONFIG_BT_CTLR_PHY_CODED=y`? S2 or S8? Or maybe I still need to do some other configurations?

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

carlescufi, 2020-04-21:
@BOZHENG001 I think it makes sense to be able to configure this from the application's perspective. Let us hear from others to see how we can add this functionality.

joerchan, 2020-04-22:
Makes sense. I have had similar requests already and have a branch ready. I can submit a PR soon.

PR #24618 (2020-05-07): Bluetooth: host: Application control of data length and phy update
Add application control of initiating data length procedure and it's
parameters.
The reasons for allowing the application control over the data length:
 - Bandwidth control adjusted based on number of active connections.
 - Changing data length before switching to coded phy.
 - Applying workarounds for interoperability problems.
 - Controlling order of ATT MTU and data length procedures.

Add application control of initiating phy change procedure and it's
parameters.
The reasons for allowing the application control over the PHY [... cut]
closed 2020-05-07 by carlescufi