zephyrproject-rtos/zephyr #23997: flash sector erase fails on stm32f412

mtahirbutt, 2020-04-01:
**Describe the bug**

 I am facing a similary problem with my disco_l475_iot1 board which has stm32l475 MCU. I am using mcuboot for updatehub app. I am not able to erase flash memory partition.

A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

flash_write_protection_set() works fine but flash_erase() inside mcuboot.c is not successful.
 I checked carefully all the partition sizes they are perfectly according to DTS file. Now I tried to see OPTCR register but could not find it.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

I should see successful erase and the method flash_erase should return 0 on successful return.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

I am working with updatehub app but not able to achieve that because of this error.

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: (e.g. Linux,)
 - Toolchain (e.g Zephyr 2.2.99, SDK,0.11.2.)
 - 

**Additional context**
Add any other context about the problem here.

nandojve, 2020-04-01:
It may be related to #18263

erwango, 2020-04-02:
^^@ABOSTM

mtahirbutt, 2020-04-02:
Hi @erwango, I tried with zephyr 2.2.0, there I tried flash page_erase 200 and that worked. so I am switching to zephyr 2.2.0 and will now try updatehub app.

erwango, 2020-04-03:
@mtahirbutt out of curiosity, on which release did you face the issue ?

otavio, 2020-04-03:
@mtahirbutt to be sure, this works on 2.2 and fails on master, right?

PR #24059 (2020-04-07): drivers: flash: stm32: Use correct driver compatible
drivers: flash: stm32: Use correct driver compatible

Use valid stm32 flash driver compatibles instead of flash
area compatible.
For 'write_block_size' property, use reference to soc-nv-flash.

fixes #23997
commit e4589b86 (2020-04-07): drivers: flash: stm32: Use correct driver compatible

Use valid stm32 flash driver compatibles instead of flash
area compatible.
For 'write_block_size' property, use reference to soc-nv-flash.

fixes #23997
closed 2020-04-07 by nashif