zephyrproject-rtos/zephyr #24008: Build failure on intel_s1000_crb board.

chen-png, 2020-04-02:
**Describe**
Couldn't build test cases on i=intel_s1000_crb board, it would report a build error.

**To Reproduce**
Steps to reproduce the behavior:
1.  west build -b intel_s1000_crb -p auto samples/hello_world/
2.  See error

**Screenshots or console output**
In file included from /home/chenpeng/zephyrproject/zephyr/kernel/queue.c:20:
..**/kernel/include/ksched.h:17:43: error: too few arguments provided to function-like macro invocation
             >= K_HIGHEST_APPLICATION_THREAD_PRIO);**
                                                 ^
**../include/toolchain/common.h:144:9: note: macro 'BUILD_ASSERT' defined here
#define BUILD_ASSERT(EXPR, MSG)**                          \
        ^
In file included from /home/chenpeng/zephyrproject/zephyr/kernel/queue.c:20:
../kernel/include/ksched.h:16:1: error: type specifier missing, defaults to 'int' [-Werror,-Wimplicit-int]
BUILD_ASSERT(K_LOWEST_APPLICATION_THREAD_PRIO
^
2 errors generated.

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.11.1
 - Commit ID: 1134edd6ad1d

dcpleung, 2020-04-02:
I have no problem building on master (commit 2e879fae4841b28c5ccb0fef4951fb1fa6efb359, two commits ahead of 1134edd), and Zephyr SDK 0.11.2.

chen-png, 2020-04-03:
i reset to 2e879fa and  updated sdk ot 0.11.2, it still reported this build error.
i will check my test environment firstly.

chen-png, 2020-04-07:
it still reported the error, is there any config i missed?
**../include/toolchain/common.h:144:9: note: macro 'BUILD_ASSERT' defined here
#define BUILD_ASSERT(EXPR, MSG)**

i reverted my repo to commit 2b0e9a3d, it was ok to build tests cases.

dcpleung, 2020-04-07:
I can build with latest master (commit 093fec62e7bddb6c1f6ad8a204d992759ef8d7ac) without any issues. This is on that commit *without* any modifications to the tree. I can't debug this until I can reproduce this issue locally.

nashif, 2020-04-07:
@chen-png  what is the ZEPHYR_TOOLCHAIN_VARIANT you are using? zephyr or xcc?

chen-png, 2020-04-08:
> zephyr or xcc?

i used the xcc and validated a test before, it was ok to run tests on intel_s1000 board using my environment.

export XTENSA_TOOLCHAIN_PATH=/home/chenpeng/xtensa
export ZEPHYR_TOOLCHAIN_VARIANT=xcc
export TOOLCHAIN_VER=RI-2018.0-linux
export XTENSA_CORE=X6H3SUE_RI_2018_0
export XTENSA_SYSTEM=${XTENSA_TOOLCHAIN_PATH}/XtDevTools/install/tools/RI-2018.0-linux/XtensaTools/config/
export XTENSA_BUILD_PATHS=${XTENSA_TOOLCHAIN_PATH}/XtDevTools/install/builds/
export XTENSA_OCD_PATH=/opt/tensilica/xocd-12.0.4

nashif, 2020-04-10:
@chen-png  then please say it is xcc, your bug report says:
```
Toolchain: zephyr-sdk-0.11.1
```

this is misleading

chen-png, 2020-04-11:
> please say it is xcc

updated it.

PR #24273 (2020-04-11): toolchain: Fix warning about too few arguments on BUILD_ASSERT()
Commit c408fa88a36a595306e96d9b06422e8d1c051c57 introduced changes
to the BUILD_ASSERT() macro so the MSG argument is optional.
However, in include/toolchain/common.h, the BUILD_ASSERT()
definition has not made the MSG argument as optional which results
in build errors complaing about too few arguments. Fix by adding
some dots there.

This is observed when using XCC (based on Clang 3.9).

Fixes #24008
commit 04a3d990 (2020-04-11): toolchain: Fix warning about too few arguments on BUILD_ASSERT()

Commit c408fa88a36a595306e96d9b06422e8d1c051c57 introduced changes
to the BUILD_ASSERT() macro so the MSG argument is optional.
However, in include/toolchain/common.h, the BUILD_ASSERT()
definition has not made the MSG argument as optional which results
in build errors complaing about too few arguments. Fix by adding
some dots there.

This is observed when using XCC (based on Clang 3.9).

Fixes #24008
closed 2020-04-11 by nashif