zephyrproject-rtos/zephyr #24009: Bluetooth: Mesh: Friend node not cache ALL_Node Address or different app_idx

LingaoMeng, 2020-04-02:
1. set up there nodes, call publish node, friend node ,lpn node.

All node share single network key, but friend & lpn node with different application key, 
the lpn node has same application key will publish node.

when publish node periodic send model message with the dst = 0xffff, lpn node not receive all of them.

LingaoMeng, 2020-04-02:
https://github.com/zephyrproject-rtos/zephyr/blob/c98652b42f12f07736259788ba067d8987387cb9/subsys/bluetooth/mesh/transport.c#L905-L914 and sdu_recv_seg

should be change as return 0 ?

LingaoMeng, 2020-04-02:
https://github.com/zephyrproject-rtos/zephyr/blob/c98652b42f12f07736259788ba067d8987387cb9/subsys/bluetooth/mesh/friend.c#L1387-L1411 should add judgement for ALL_NODE_ADDRESS ?

jhedberg, 2020-04-02:
I remember seeing some discussion in the Mesh WG about this, with the conclusion that LPNs need to explicitly subscribe to these addresses, so based on that it's not a bug. @trond-snekvik can you confirm?

trond-snekvik, 2020-04-02:
I remember this too, according to spec, the LPN has to explicitly subscribe to 0xffff. There's no public API for doing this in zephyr though, as it's controlled by the config server, which won't accept 0xffff. Perhaps it could be compile time option to automatically subscribe to 0xffff when the friendship is established?

LingaoMeng, 2020-04-02:
well, it's still another problem, friend node has different `appkey `with lpn node , but currently call `sud_recv` return `-EINVAL` when local node unable find `appkey`,
so, even the friend node known its dst is lpn node,  and not store this to friend queue.

jhedberg, 2020-04-02:
@LingaoMeng let's not mix multiple issues into a single GitHub issue. Open a new one since I already categorised this based on the 0xffff address concern.

PR #24327 (2020-04-19): Bluetooth: Mesh: Fix lpn groups add when node reset
Add lpn groups when model sub restore after reset.

Add auto sub `ALL_NODE_ADDR` when provisioned.

Remove all groups when node unprovisoned.

Fixes: #24311
Fixes: #24009
commit 2e4eab5e (2020-04-19): Bluetooth: Mesh: Add option for lpn auto sub all-nodes

Add option for low power node automatically subscribe
all-nodes-address to friend sub list.

Fixes: #24009
closed 2020-04-19 by jhedberg