zephyrproject-rtos/zephyr #24028: [Coverity CID :209381] Unrecoverable parse warning in include/bluetooth/bluetooth.h

zephyrbot, 2020-04-02:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/29c0c08fd4abc34aae4e1bf579cc506eaabef428/include/bluetooth/bluetooth.h#L729

Category: Parse warnings
Function: ``
Component: Bluetooth
CID: [209381](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=209381)

Details:

```
723         /** Disable scan on 1M phy.
724          *
725          *  @note Requires @ref BT_LE_SCAN_OPT_CODED.
726          */
727         BT_LE_SCAN_OPT_NO_1M = BIT(3),
728    
>>>     CID 209381:  Parse warnings  (PARSE_ERROR)
>>>     {CovLStr{v1}{expected a "}"}}
729         BT_LE_SCAN_FILTER_DUPLICATE __deprecated =
730             BT_LE_SCAN_OPT_FILTER_DUPLICATE,
731         BT_LE_SCAN_FILTER_WHITELIST __deprecated =
732             BT_LE_SCAN_OPT_FILTER_WHITELIST,
733     };
734    
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

joerchan, 2020-04-02:
Is the issue here that there is a line break after the assignment?
I haven't seen any issues with this line: `BT_SECURITY_NONE   __deprecated = BT_SECURITY_L0,`

jhedberg, 2020-04-02:
I would just clasify this as a Coverity bug, i.e. nothing to do with the code. Even the category speaks for that: "Parse warning". The code is clearly valid so this is just Coverity's shortcoming.

nashif, 2020-08-26:
dismissed

closed 2020-08-26 by nashif