zephyrproject-rtos/zephyr #24031: [Coverity CID :209378] Pointless string comparison in tests/lib/devicetree/src/main.c

zephyrbot, 2020-04-02:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/29c0c08fd4abc34aae4e1bf579cc506eaabef428/tests/lib/devicetree/src/main.c#L1051

Category: Incorrect expression
Function: `test_cs_gpios`
Component: Tests
CID: [209378](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=209378)

Details:

```
1045         zassert_equal(DT_SPI_HAS_CS_GPIOS(TEST_SPI_NO_CS), 0, "unexpected cs");
1046         zassert_equal(DT_SPI_NUM_CS_GPIOS(TEST_SPI_NO_CS), 0, "wrong no. of cs");
1047    
1048         zassert_equal(DT_SPI_HAS_CS_GPIOS(TEST_SPI), 1, "missing cs");
1049         zassert_equal(DT_SPI_NUM_CS_GPIOS(TEST_SPI), 3, "wrong no. of cs");
1050    
>>>     CID 209378:  Incorrect expression  (CONSTANT_EXPRESSION_RESULT)
>>>     "strcmp("TEST_GPIO_1", "TEST_GPIO_1")" is always 0 because ""TEST_GPIO_1"" is compared against itself.
1051         zassert_true(!strcmp(DT_SPI_DEV_CS_GPIOS_LABEL(TEST_SPI_DEV_0),
1052                      "TEST_GPIO_1"),
1053                  "dev 0 cs gpio name");
1054         zassert_equal(DT_SPI_DEV_CS_GPIOS_PIN(TEST_SPI_DEV_0), 0x10,
1055                   "dev 0 cs gpio pin");
1056         zassert_equal(DT_SPI_DEV_CS_GPIOS_FLAGS(TEST_SPI_DEV_0), 0x20,
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

galak, 2020-04-02:
False positive, this is what the test is trying to do.

closed 2020-04-02 by galak

nashif, 2020-04-02:
Did you mark those in coverity as false-positive?