zephyrproject-rtos/zephyr #24051: double to sensor_val

GeorgeGkinis, 2020-04-02:
**Is your feature request related to a problem? Please describe.**

I want to convert a double to sensor_val

**Describe the solution you'd like**
I would like a function **double_to_sensor_value** the way **sensor_value_to_double** exists.

**Describe alternatives you've considered**
Googled allot.

**Additional context**
None.

JonBruchim, 2020-10-04:
would like to get assigned to this issue

saltyJeff, 2020-11-10:
Any progress made on this? I'm currently using this function as an inline: https://repl.it/@JeffersonLee/zephyr-double-to-sensor-val#main.c

I included the test cases from: https://docs.zephyrproject.org/latest/reference/peripherals/sensor.html#c.sensor_value

henrikbrixandersen, 2021-07-30:
A helper function `sensor_value_from_double(struct sensor_value *, double)` was added in commit 3c36cd4242ae94971ebe4d6dfcc6ade6ea9a6cb8.

Closing.

closed 2021-07-30 by henrikbrixandersen