zephyrproject-rtos/zephyr #24055: Add support for openocd on stm32g0 and stm32g4 targets

erwango, 2020-04-03:
**Is your enhancement proposal related to a problem? Please describe.**
stm32g0 and stm32g4 soc support has recently been added on openocd.
openocd support can now be enabled on these targets when flashing  debugging

**Describe the solution you'd like**
Identify openocd SHA1 that enable G4/G0 support.
Enable openocd support on zephyr G4 and G0 supported platforms.
Update board docs
Check everything is fine flash/debug
Request needed change on https://github.com/zephyrproject-rtos/openocd
Close related issues (https://github.com/zephyrproject-rtos/openocd/pull/11)

**Additional context**
Will avoid need to update pyocd local version to support G4 packages.

FRASTM, 2020-04-03:
Actually, I can see the PR https://github.com/zephyrproject-rtos/openocd/pull/11
for G4 but no G0 support

erwango, 2020-04-03:
@FRASTM 
This is on openocd main repo: http://openocd.zylin.com/#/c/4807/
zephyrproject-rtos/openocd is a fork that needs to be updated

FRASTM, 2020-04-07:
### To install and use your local openocd:

1. install tools needed to compile OpenOCD
`sudo apt-get install git autoconf libtool make pkg-config libusb-1.0-0 libusb-1.0-0-dev`

2. Download the latest source code for OpenOCD with
`git clone http://openocd.zylin.com/openocd`

3. Change into the code directory and run the bootstrapper with:
`cd openocd-code`
`./bootstrap`

4. Run `make` and `sudo make install`

5. You are now ready to Flash your board with 
`west flash --openocd /usr/local/bin/openocd
`

closed 2020-04-15 by ioannisg