zephyrproject-rtos/zephyr #24061: PVS-Studio: Checking the Code of Zephyr

AndreyKarpov, 2020-04-03:
While working on PVS-Studio's integration into PlatformIO, we kept in touch with the PlatformIO team, and they suggested that we check Zephyr, a project from the embedded software world. We liked the idea, and that's how this article appeared.

Here you can check out the article and fix several found errors: [Checking the Code of Zephyr](https://www.viva64.com/en/b/0721/).

However, perhaps you should know that we were not intended to find as many errors as possible. Therefore, if you're interested in the topic, I suggest checking the project yourself more meticulously. Since the project is open-source and available on GitHub, you can use a [free PVS-Studio licensing](https://www.viva64.com/en/b/0614/) option.

nashif, 2020-04-03:
@AndreyKarpov  Thank you for the article, I have downloaded the tool and experimented with it a bit, very easy to use and was straightforward to setup and get results. 

We will continue evaluating PVS-Studio for more advanced usage and integration with our development processes.

closed 2023-08-14 by nashif

GrixaYrev, 2025-03-26:
@nashif, hello! Can you help me to test pvs-studio?
I add new target to main CMakeLists.txt, but i don't know what target i should set to ANALYZE. I tried **app** and **zephyr**, both not full.
```
pvs_studio_add_target(TARGET analyze ALL
                      OUTPUT FORMAT errorfile
                      RECURSIVE ANALYZE app
                      LOG target.err)
```