zephyrproject-rtos/zephyr #24067: cross-platform inconsistency in I2C bus speeds

pabigot, 2020-04-03:
The [I2C bus specification](http://cache.nxp.com/documents/user_manual/UM10204.pdf) in section 5 specifies these named speeds:

* Standard up to 100 kHz
* Fast up to 400 kHz
* Fast+ up to 1 MHz
* High up to 3.4 MHz
* Ultra up to 5 MHz (unidirectional only)

Evaluation in https://github.com/zephyrproject-rtos/zephyr/pull/23971#pullrequestreview-387200780 plus code review show that NXP (mcux, mcux_lpi2c, imx) and SiLabs (gecko) I2C drivers use 1 MHz for Fast, and do not implement Fast+.

The interpretation of named I2C bus speeds in Zephyr should be (a) documented and (b) consistent.

carlescufi, 2020-04-07:
@mnkp and @MaureenHelm could you send a patch to correct Fast so it becomes Fast+?

MaureenHelm, 2020-04-07:
> @mnkp and @MaureenHelm could you send a patch to correct Fast so it becomes Fast+?

yes

PR #24159 (2020-04-20): drivers: i2c: Fix incorrect fast and fast+ mode bus speeds
Fixes the nxp and silabs i2c drivers to decode fast and fast+ mode bus
speeds as 400 kHz and 1 MHz respectively.

~Depends on #23119~
Fixes #24067
closed 2020-04-20 by carlescufi