zephyrproject-rtos/zephyr #24072: tests/kernel/timer/timer_api failed on nucleo_stm32l152re board

FRASTM, 2020-04-03:
**Describe the bug**
running the tests/kernel/timer/timer_api  on nucleo_stm32l152re board, it fails on test case test_timer_remaining

similar to issue https://github.com/zephyrproject-rtos/zephyr/issues/24001

**To Reproduce**
Steps to reproduce the behavior:

    west build -b nucleo_l152re -p auto tests/kernel/timer/timer_api/
    west flash
    See error

**Screenshots or console output**

```
=================================================================== 
starting test - test_timer_remaining                                                                                                                                                                        

Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/src/main.c:546: test_timer_remaining: rem_ticks <= dur_ticks / 2 is false                                                           
                                                                                                                                                                                                            
FAIL - test_timer_remaining  
=================================================================== 

```
all other tests: PASS

**Environment (please complete the following information):**

 - Commit SHA or Version used
8d5b998582d747064d0624d9bf9649b999aa60f8

FRASTM, 2020-04-03:
note that tests/kernel/timer/timer_api also fails when running on the nucleo_g071rb board

- testcase test_timer_remaining fails
- and also testcase test_timeout_abs fails :

===================================================================      
starting test - test_timeout_abs                                         
                                                                         
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/e
                                                                         
FAIL - test_timeout_abs                                                  
===================================================================

dcpleung, 2020-04-03:
#24080 should fix the fail `test_timer_remaining()` test.

FRASTM, 2020-04-06:
Well, https://github.com/zephyrproject-rtos/zephyr/pull/24080, fixes for the nucleo_L152

However, testing with the stm32_lptimer as tick tick timer (on nucleo_l4r5zi) It seems that the same fix is expected on the previous assert of this testcase : 

zassert_true(rem_ticks <= (dur_ticks / 2) **+ 1** , NULL);

FRASTM, 2020-04-23:
PR Fixups to timer_api test #24080 is now merged

closed 2020-04-23 by FRASTM