zephyrproject-rtos/zephyr #24076: UARTE high current consumption

rerickson1, 2020-04-03:
**Describe the bug**
If the UARTE peripheral is used on the NRF52840, it prevents the 840 from entering low power mode.

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout example project from here: https://github.com/rerickson1/nrf52840_lpm
2. build for nrf52840_pca10056
3. Power pca10056 from P21 with 3.3V
4. Flash the example and the board will go to sleep properly (~10 uA)
5. Uncomment line 1 in `prj.conf` and replace line 2 in `nrf52840_pca10056.overlay` with `compatible = "nordic,nrf-uarte";`
6. Rebuild and flash the example onto the board.
7. Now the board cannot go to sleep properly.  The min current is ~500 uA.

**Expected behavior**
UARTE peripheral should shutdown properly when calling `device_set_power_state(uart_dev, DEVICE_PM_OFF_STATE, NULL,NULL);`

**Impact**
showstopper

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: gcc-arm-none-eabi-8-2019-q3-update
 - Commit SHA or Version used: v1.14.1

**Additional context**
My pca10056 has silicon rev AAC0.  I have tried [the following work around](https://devzone.nordicsemi.com/f/nordic-q-a/26030/how-to-reach-nrf52840-uarte-current-supply-specification/102605#102605) after setting the power state to off and it fixes the problem.  The issue is that resetting the UART peripheral clears all registers and it will not work if you set the power state back to on.

nashif, 2020-04-04:
@carlescufi @pabigot  fyi

carlescufi, 2020-04-07:
@pabigot @Mierunski and @anangl  this is in 1.14.1, is there something we can backport to improve this?

endian-benjamin, 2020-04-07:
For the record, we have a big project running 1.14.1 in which we use UARTE on UART_1 with nrf52840 and with strict consumption requirements. Did you try disabling uart irq:s before `device_set_power_state()`? That seemed necessary for us.

rerickson1, 2020-04-07:
> For the record, we have a big project running 1.14.1 in which we use UARTE on UART_1 with nrf52840 and with strict consumption requirements. Did you try disabling uart irq:s before `device_set_power_state()`? That seemed necessary for us.

In my example interrupt support for the UART is disabled, so that shouldn't be needed.  Could you provide a code snip-it of what you did?

endian-benjamin, 2020-04-07:
Calling `uart_irq_rx_disable()` on the device before setting power state was basically it (TX is done synchronously so we know those are not enabled). We use UARTE on both UARTs and we've done pretty extensive measurements. CPU definitely enters low power.

Perhaps interrupt driven UART is required for this to work with UARTE.

endian-benjamin, 2020-04-07:
(by "this" I mean power management ofc)

rerickson1, 2020-04-07:
@endian-benjamin  I enabled `CONFIG_UART_INTERRUPT_DRIVEN=y` and tried to disable the RX IRQ and the result is the same.  I am only getting to ~700uA.  Here's how I did it:
```
static void shutdown_console_uart(void)
{
	struct device *uart_dev;
	uart_dev = device_get_binding(CONFIG_UART_CONSOLE_ON_DEV_NAME);

#ifdef CONFIG_UART_INTERRUPT_DRIVEN
	uart_irq_rx_disable(uart_dev);
#endif

	int rc = device_set_power_state(uart_dev, DEVICE_PM_OFF_STATE, NULL,
					NULL);

	// *(volatile uint32_t *)0x40002FFC = 0;
	// *(volatile uint32_t *)0x40002FFC;
	// *(volatile uint32_t *)0x40002FFC = 1;
	if (rc) {
		printk("Error disabling console UART peripheral (%d)\n", rc);
	}
}

/*
 * Application defined function for doing any target specific operations
 * for low power entry.
 */
void sys_pm_notify_power_state_entry(enum power_states state)
{
	switch (state) {
	case SYS_POWER_STATE_SLEEP_1:
		printk("--> Entering SYS_POWER_STATE_SLEEP_1 state.\n");
		shutdown_console_uart();
		k_cpu_idle();
		break;
	default:
		printk("Unsupported power state %u\n", state);
		break;
	}
}
```

rerickson1, 2020-04-07:
I pushed updates to my example for testing.

endian-benjamin, 2020-04-08:
We actually have a currently unresolved bug where our device doesn't enter low power mode after a hard reset, for some reason... but it starts working after a soft reset (watchdog trigger or calling reset() for instance).

I'd also try checking if separating the printk calls from the suspend calls to see if that helps, i.e. make sure the bytes have actualy left the uart hw buffer BEFORE you suspend the device, to make sure this isn't what's keeping the UART from getting suspended properly.

rerickson1, 2020-04-08:
> I'd also try checking if separating the printk calls from the suspend calls to see if that helps, i.e. make sure the bytes have actualy left the uart hw buffer BEFORE you suspend the device, to make sure this isn't what's keeping the UART from getting suspended properly.

nice thought.  I removed the `printk` calls from `sys_pm_notify_power_state_entry`.  No luck... Still leaving the HFCLK active (~500 uA).

rerickson1, 2020-04-08:
Kept my example exactly the same and switched to Zephyr 2.2.0 and everything works fine there.  So something changed with the UARTE behavior between 1.14.1 and 2.2.0.

rerickson1, 2020-04-08:
Looks like this commit is what could fix the issue: 55e2014d995a4a6b73cc1d967bdd4fd452ceb03e. This commit is not included in 1.14.1
@endian-benjamin looks like you were the author!

rerickson1, 2020-04-09:
I can confirm that commit fixes shutting down the UART properly.  There is now an issue when we come back out of sleep:
```
***** Booting Zephyr OS zephyr-v1.14.1-1-g85dfcab1ec32 *****

*** Power Management Demo on nrf52840_pca10056 ***

App busy waiting for 5 seconds
App sleeping for 15 seconds
--> Entering SYS_POWER_STATE_SLEEP_1 state.
<-- Exited SYS_POWER_STATE_SLEEP_1 state.
***** MPU FAULT *****
  Instruction Access Violation
***** Hardware exception *****
Current thread ID = 0x200000f4
Faulting instruction address = 0x200024e0
Fatal fault in essential thread! Spinning...
```

I'm not sure what would cause this...

endian-benjamin, 2020-04-09:
@rerickson1 Yeah I sorta assumed that commit was backported to v1.14 - we obviously have it in our downstream.

endian-benjamin, 2020-04-09:
GDB is your best bet, step around a bit and also try `disassemble` on/around the culprit address

rerickson1, 2020-04-09:
Thanks, I'll poke around.  Im am guessing there may be some other fixes I need along with that commit. 
 Unfortunately it looks like there are a lot of NRF UART fixes that didn't make it into 1.14.1 even though they were committed months before the release.

carlescufi, 2020-04-09:
@rerickson1 the issue is that we only backport critical security fixes or whatever the community explictly asks to be backported. If you want something backported, the safest way is to send a PR against the v1.14-branch and add it to the Backports GitHub project.

rerickson1, 2020-04-09:
Thanks @carlescufi.  I am testing out backporting the fixes now.

rerickson1, 2020-04-09:
For clarity, branch v1.14-branch already contains a fix that resolves the high current consumption of UARTE when powering it off.  The commit that fixes in on that branch is a2e7b0ddf4b76875f503b7e29c02f56ba8248885.  the 1.14 branch is still missing several critical fixes and the 1.14.1 release tag is missing several fixes along with the one to fix high current consumption.  PR #24245 aims to bring all UART/UARTE driver fixes to 1.14.x

rerickson1, 2020-07-08:
Resolved with #24245

closed 2020-07-08 by rerickson1