zephyrproject-rtos/zephyr #24081: le_adv_ext_report is not generating an HCI event

lagosantol, 2020-04-03:
**Describe the bug**
The le_adv_ext_report  is only logging, but is not changing the content of the buf parameter.
As a consequence, no BT_HCI_EVT_LE_EXT_ADVERTISING_REPORT is sent to the host.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_nrf52840dk_nrf52840
3. make
4. flash 
5. plug the board using USB on a RPi
6. bt_attach hci1
7. btmon on hci1
8. hcitool lescan on hci1

Use a device that advertises using extended mode (in PHY 1MBPS) 
The device is not seen by lescan.
No extended advertisement event appears in the btmon log.

**Expected behavior**
A clear and concise description of what you expected to happen.
There should be an event received by the host over HCI.
Like there is one for advertisement reports.
It's probably because no call is made to meta_evt, using BT_HCI_EVT_LE_EXT_ADVERTISING_REPORT as the type.
And the event is not filled (the bt_hci_evt_le_ext_advertising_info is never used).
As a consequence, the event buf is not null, but it's of zero length, so it's ignored later on and never sent to the host.

**Impact**
Showstopper, we want to use zephyr for a scanning device that has to support extended advertising (and CODED_PHY).

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: Linux, Ubuntu 18.04 LTS
 - Toolchain Zephyr SDK
 - Commit SHA: b9d8884c253c9658cab2954e3d5d704d86f1584c

**Additional context**
Add any other context about the problem here.

carlescufi, 2020-04-07:
@lagosantol the Zephyr LL doesn't support Advertising Extensions yet, we are in the process of adding that. I assume this is the issue.
@joerchan please double-check and close the issue if this is the case

joerchan, 2020-04-07:
It's work in progress. I'll close this.

closed 2020-04-07 by joerchan

lagosantol, 2020-04-08:
Can you please point me to the issue tracking the completion process for advertising extensions support ? So I can track progress, and evaluate if we should help here ?
Is it correct to say that with this support, you can't have a device that does CODED_PHY only ?
(for adverstising first, and then connected mode)

joerchan, 2020-04-08:
@lagosantol This work is tracked here: https://github.com/zephyrproject-rtos/zephyr/projects/21
There might not even be an issue for extended advertising support in the scanner yet.

It is correct that advertising extension is needed to only be on CODED Phy.

carlescufi, 2020-04-08:
@lagosantol it is @cvinayak who is working on the controller support for Extended Advertising. He can give you some more information on the status, but the focus right now is non-connectable, non-scannable extended advertising first, then the rest.

PR #22237 is the first step towards Extended Adv in the controller

Also feel free to join the #bluetooth channel in [slack](https://tinyurl.com/y5glwylp) to coordinate work if you want to help.