zephyrproject-rtos/zephyr #24092: Unable to change recv() buffer size in frdm_k64f board.

Navin-Sankar, 2020-04-04:
I am trying to download a file from http server with file size of 11kb with frdm_k64f board. I am using the http_get socket sample as my reference. My intention is to receive a file with response buffer size as 1KB per receive. But it receives only 640 bytes per every receive. Sample application and configuration file attached.

#include <stdio.h>
#include <stdlib.h>

#include <net/socket.h>
#include <kernel.h>
#include "dhcp.h"

#define HTTP_HOST "192.168.1.109"
#define HTTP_PORT  80
#define HTTP_PATH "/file.txt"

#define SSTRLEN(s) (sizeof(s) - 1)
#define CHECK(r) { if (r == -1) { printf("Error: " #r "\n"); exit(1); } }
#define REQUEST "GET " HTTP_PATH " HTTP/1.0\r\nHost: " HTTP_HOST "\r\n\r\n"

static char response[1024];

void main(void)
{
	int sock;
	struct sockaddr_in serv_addr;
	app_dhcpv4_startup();
        serv_addr.sin_family = AF_INET;
        serv_addr.sin_port = htons(HTTP_PORT);
	sock = socket(AF_INET, SOCK_STREAM, IPPROTO_TCP);
	if (sock < 0) {
		printk("Failed to create TCP socket \n");
	}
        if(inet_pton(AF_INET, HTTP_HOST, &serv_addr.sin_addr)<=0)
        {
                printk("Invalid address/ Address not supported \n");
        }
        CHECK(sock);
	printf("sock = %d\n", sock);
	CHECK(connect(sock, (struct sockaddr *)&serv_addr, sizeof(serv_addr)));
	CHECK(send(sock, REQUEST, SSTRLEN(REQUEST), 0));
	printf("Response:\n\n");
	while (1) {
		int len = recv(sock, response, sizeof(response), 0);
		if (len < 0) {
			printf("Error reading response\n");
			return;
		}
		if (len == 0) {
			break;
		}
		printk("Response len: %d \n", len);
//		response[len] = 0;
//		printf("%s", response);
	}
	printf("\n");
	(void)close(sock);
}

prj.conf
//General config
CONFIG_NEWLIB_LIBC=y

// Networking config
CONFIG_NETWORKING=y
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n
CONFIG_NET_TCP=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y

CONFIG_DNS_RESOLVER=y
CONFIG_DNS_SERVER_IP_ADDRESSES=y
CONFIG_DNS_SERVER1="8.8.8.8"

CONFIG_BOOTLOADER_MCUBOOT=y

// Network driver config
CONFIG_TEST_RANDOM_GENERATOR=y

CONFIG_NET_LOG=y

CONFIG_BUILD_OUTPUT_HEX=y

CONFIG_NET_DHCPV4=y
CONFIG_NET_MGMT=y
CONFIG_NET_MGMT_EVENT=y

CONFIG_MAIN_STACK_SIZE=4096

Navin-Sankar, 2020-04-04:
client = 192.168.1.107 & server = 192.168.1.109.
The tcp window size for frdm_k64f client is 1280. Is the reason for (1280/2) = 640 bytes per receive? 

[image]

jukkar, 2020-04-06:
Default MSS (Maximum Segment Size) for IPv4 is 536 bytes. Default Ethernet + IPv4 + TCP headers are 14 + 20 + 20 = 54 bytes. So by default the packet length is 590 bytes if nothing else is negotiated. This is what I saw in my tests with qemu_x86 and e1000 with echo-server/client sample apps.
For you, the MSS seems to be a big larger as the packet length is 694 (so 54 header and 640 data).

pfalcon, 2020-06-30:
In addition to what @jukkar said: this feature is called "short read" (correspondingly, there's "short write"). With POSIX, any read() or similar (recv(), etc.) operation may return **less** bytes than requested. Just the same as any read() operation may return 0 (meaning EOF/connection closed) or negative value (meaning error). Dealing with all of these is up to application programmer. In particular, if you received less bytes than expected, you should call read()/recv again (and multiple times as needed), adjusting the receive buffer pointer and require length accordingly. Similarly, for write()/send().

pfalcon, 2020-06-30:
Hopefully, this answers the question.

@jukkar, @MaureenHelm: I'd suggest to close this ticket. Not doing this right away, in case there would be further questions.

closed 2020-07-01 by MaureenHelm