zephyrproject-rtos/zephyr #24098: drivers: flash: stm32_flash: usage fault

rckTom, 2020-04-05:
Hi,

I found a bug in the flash driver of the stm32. In most cases or
configurations, the bug won't show up. It's an issue of flash_stm32.c in combination with
its device-tree configuration. stm32f3.dtsi states:

```
[...]
                flash-controller@40022000 {
                        compatible = "st,stm32f3-flash-controller";
                        label = "FLASH_CTRL";
                        reg = <0x40022000 0x400>;
                        interrupts = <4 0>;

                        #address-cells = <1>;
                        #size-cells = <1>;

                        flash0: flash@8000000 {
                                compatible = "soc-nv-flash";
                                label = "FLASH_STM32";

                                write-block-size = <2>;
                                erase-block-size = <2048>;
                        };
                };
[...]
```

Later, the control registers are loaded in flash_stm32.c:256,
flash_stm32_write_protection() tries to load the base address of the
control registers:

```
#define DT_DRV_COMPAT soc_nv_flash
[...]

static struct flash_stm32_priv flash_data = {
        .regs = (FLASH_TypeDef *) DT_INST_REG_ADDR(0),

```
```
FLASH_TypeDef *regs = FLASH_STM32_REGS(dev);
```

regs will point to 0x8000000, which is surely not the base address of
the control registers. This is typically the vector table base address,
and in most cases, by luck, only content of registers is read, as
further checks of the registers evaluate 'gratefully'. However, I found
this bug as the content of my vector table triggers the issues that
happens later in flash_stm32.c:272:

```
if (regs->CR & FLASH_CR_LOCK) {
        regs->KEYR = FLASH_KEY1;  /* Bad memory access */
        regs->KEYR = FLASH_KEY2;
}
```

The condition is true, and regs->KEYR writes, trying to access the vector
table and triggers a fault.

Big thanks to @rralf for helping debugging this issue.

CC:
@galak
@superna9999

tagunil, 2020-04-05:
@rckTom, does PR #24059 fix these faults?

rckTom, 2020-04-05:
Yes, PR #24059 fixes the fault.

rralf, 2020-04-05:
Tom, you then might want to add your Tested-by to that PR. At least we had a really funny remote-ssh-screen debugging session. ;-)

closed 2020-04-07 by nashif