zephyrproject-rtos/zephyr #24101: Bluetooth: Mesh: Transport Segment send failed lead to seg_tx un-free

LingaoMeng, 2020-04-05:
https://github.com/zephyrproject-rtos/zephyr/blob/7ab754d9973e7b7481658bd937e186362f603f70/subsys/bluetooth/mesh/transport.c#L400-L414

In `seg_tx_send_unacked` func, if the first `segment `unable send, this function will be terminated and not set any `retransmit` timer, this will be occupy `seg_tx` object **forever**.

Another problem:
https://github.com/zephyrproject-rtos/zephyr/blob/7ab754d9973e7b7481658bd937e186362f603f70/subsys/bluetooth/mesh/transport.c#L313-L320

if other segment(seg_o) unable send duce to unsufferent buffers, this will be retransmit immediately, seg_o already clear to zero in `seg_tx_send_unacked`, so previous successful sended segment will be sending again.

LingaoMeng, 2020-04-05:
@trond-snekvik , well,  i'm look  new transport.c in mesh dir,  https://github.com/zephyrproject-rtos/zephyr/blob/7ab754d9973e7b7481658bd937e186362f603f70/subsys/bluetooth/mesh/transport.c#L75-L80
https://github.com/zephyrproject-rtos/zephyr/blob/7ab754d9973e7b7481658bd937e186362f603f70/subsys/bluetooth/mesh/transport.c#L101-L105

`Segment array `define in both TX and RX struct, It's going to be terrifying if it supports very large data(32 Segment for example) throughput, can be replaced by `sys_slist_t` ? and currently segment for RX when received completed will be copy and copy again for decryption (this is not a good way).

I'm very confused about the current way, but I can't solve it. I hope it can help you.

LingaoMeng, 2020-04-05:
another problem:
 https://github.com/zephyrproject-rtos/zephyr/blob/7ab754d9973e7b7481658bd937e186362f603f70/subsys/bluetooth/mesh/transport.c#L1483-L1490
while in `seg_rx_alloc` allocate all segment, but not to check return value of `k_mem_slab_alloc`, so, if  no buf available, the `rx->seg[j]` will be set to 'NULL',
 https://github.com/zephyrproject-rtos/zephyr/blob/7ab754d9973e7b7481658bd937e186362f603f70/subsys/bluetooth/mesh/transport.c#L1676 will be cause set to zero-point, and finally to be dead.

trond-snekvik, 2020-04-06:
Okay, this is a lot of issues for one issue, I'll try to go through them one by one.

I looked through your PR, and the first issue and its fix looks good to me, well spotted!

> Segment array define in both TX and RX struct, It's going to be terrifying if it supports very large data(32 Segment for example) throughput, can be replaced by sys_slist_t ? 

This would just move the memory to the slabs though, which would have to fit a `sys_snode_t` and a segment index. I see that this might balance out if the number of segment contexts increases without proportionately increasing the number of slabs, but in my opinion, a linked list would not necessarily be an improvement memory wise. It would be nice to reduce the memory requirement for each segmented message context though, I agree it's a problem.

> currently segment for RX when received completed will be copy and copy again for decryption (this is not a good way).

I have to disagree somewhat, this is solving a different problem: The only other way to do this in a thread safe manner would be to allocate another temporary buffer of 382 bytes on stack to assemble the encrypted packet. This makes the total stack size contribution for this function > 800 bytes. I'd rather reduce the amount of unnecessary rebuilding here, I think. Currently, we're rebuilding for every attempted decrypt, but only a small percentage of the attempts would actually manipulate the buffer. I can make a PR to reduce this problem. Optimally, the buffer should only have to be reassembled in one of two cases: We received a devkey-encrypted packet that was encrypted with our own devkey while we're a provisioner (generally shouldn't happen), or we have two appkeys with the same 7 bit ID and the same net idx, and we tried the wrong one first, which happens very rarely. Would you agree?

> while in seg_rx_alloc allocate all segment, but not to check return value of k_mem_slab_alloc, so, if no buf available, the rx->seg[j] will be set to 'NULL',

We're checking that we have enough buffers before the loop, and the result can't change, as this function is only called from the rx thread, which is collaborative. I think your change is good though, it's more flexible.

PR #24106 (2020-04-08): Bluetooth: Mesh: Fix transport tx and rx
Fix seg_tx occupy when no segment not send
and allocate rx_seg when receive such message.

Fixes: #24101
commit 5b991aa0 (2020-04-08): Bluetooth: Mesh: Fix transport tx and rx

Fix seg_tx occupy when no segment not send
and allocate rx_seg when receive such message.

Fixes: #24101
closed 2020-04-08 by jhedberg