zephyrproject-rtos/zephyr #24103: USB Serial Number reverses bytes in hw identifier

pabigot, 2020-04-05:
Why does the default encoding of the USB serial number string descriptor reverse the byte order of the value returned by `hwinfo_get_device_id()`?  This seems wrong.

* hwid: `f7 78 63 d5 6c ae 9a 5a`
* serial number: `5A9AAE6CD56378F7`

carlescufi, 2020-04-07:
@pabigot The problem might be in `hwinfo_get_device_id()`, which seems to not reverse bytes. See  https://github.com/zephyrproject-rtos/zephyr/issues/23444

pabigot, 2020-04-07:
There may or may not be a problem with hwinfo, but IMO it's wrong to reverse things here.  The hwinfo should be interpreted as a sequence of octets, just like a MAC address.

emob-nordic, 2020-04-08:
@pabigot What did You use to read serial number from the device and what device did you use? I am using Windows USB device viewer and for me its okay. I've checked with nrf52840_pca10056 to check serial number. The result is as expected. In my case:
```
DEVICEID[0] = 91 28 45 B6
DEVICEID[1] = DB 12 E1 CA
```

Serial number:
```
iSerialNumber:                     0x03
     English (United States)  "DB12E1CA912845B6"
```

pabigot, 2020-04-08:
I expect the serial number to match the hwinfo byte sequence.  If you look at the hwinfo byte sequence for your device I expect you'll find it to be `B6452891CAE112DB`.  So hwinfo should be fixed, and the byte reversal in USB should be removed.

sslupsky, 2020-04-08:
I looked at the nordic driver when I looked into the hwinfo issue #23444 I posted.  This is an endianness problem with hwinfo.

sslupsky, 2020-04-08:
@emob-nordic Your observation is correct.  DEVICEID[0] and DEVICEID[1] are u32_t data types so when you print them as u32_t data types, the output is printed big-endian.  However, if you copy the data using memcpy and then print a sequence of bytes as HWINFO does, the byte sequence is printed little-endian because that is the underlying endianness of the ARM Cortex M0.

pabigot, 2020-04-09:
@carlescufi This is not closed.  The USB infrastructure must be updated to not reverse the bytes in `hwinfo`, now that they are stored in the expected order.

PR #24203 (2020-04-09): drivers: hwinfo: fix endianness for sam0 and nrf
HWINFO drivers should be responsible for ensuring that
 the data structure is a sequence of bytes. That is not
 what the current sam0 and nordic drivers do. The drivers
 read the data as u32_t and then memcpy the data to a
 buffer. This ensures the data has the endianness of the
 underlying MCU, which in this case is Cortex M0 which
 is little endian.

 This commit fixes the endianness so the data can be
 interpreted as a "left to right sequence of bytes".

 Fixes #23444
Fixes #24103
PR #24244 (2020-04-10): usb: Stop reversing the hardware identifier
Now that the hardware identifier obtained with the hwinfo API is stored
in the correct order based on the platform, there is no longer the need
to reverse its bytes when populating the serial number in the
corresponding USB descriptor.

Fixes #24103
commit ba3b1e06 (2020-04-10): usb: Stop reversing the hardware identifier

Now that the hardware identifier obtained with the hwinfo API is stored
in the correct order based on the platform, there is no longer the need
to reverse its bytes when populating the serial number in the
corresponding USB descriptor.

Fixes #24103
closed 2020-04-10 by ioannisg