zephyrproject-rtos/zephyr #24104: west sign usage help is missing key information

pabigot, 2020-04-05:
```
tirzah[158]$ west sign -t imgtool --key ${SIGNING_KEY}     
usage: west sign [-h] [-d BUILD_DIR] [-f] -t {imgtool,rimage} [-p TOOL_PATH]
                 [--bin] [-B BIN] [--hex] [-H HEX]
                 [tool_opt [tool_opt ...]]
west sign: error: unexpected arguments: ['--key']
tirzah[162]$ west sign -t imgtool -- --key ${SIGNING_KEY}     
=== image configuration:
...
```

The usage help should explicitly show the requirement to add `--` before the tool options.  This probably applies to other commands.

It would also help if the documentation included a link to https://juullabs-oss.github.io/mcuboot/imgtool.html which tells you what the tool options are.  It took far to long to figure out how to put a version number into the image header.

mbolivar-nordic, 2020-04-07:
https://docs.zephyrproject.org/latest/guides/west/sign.html

closed 2020-04-07 by mbolivar-nordic

pabigot, 2020-04-07:
@mbolivar So the fact that the tool itself gives wrong instructions is not worth fixing?  Or is it not possible to fix?

pabigot, 2020-04-07:
Also my point about the documentation was that the imgtool link there is to the PyPI page which tells you nothing useful.  That does, granted, provide a multi-step path that will get you to the real imgtool documentation, but it would still be nice if Zephyr's instructions were more directly helpful.

mbolivar-nordic, 2020-04-07:
> So the fact that the tool itself gives wrong instructions is not worth fixing? Or is it not possible to fix?

I don't see where the tool is giving wrong instructions in this report. POSIX says that -- is used to unambiguously separate option arguments from non-option arguments. That is why it must be used for option arguments to imgtool that the parent script might interpret as option arguments also.

This is the same reason why you have to run `grep -r -- -pin dts` if you want to recursively search for "-pin" in a "dts" subdirectory. And yet `grep --help` does not mention that PATTERNS must come after a `--`, even though you can see this:

```
$ grep -r -pin dts
grep: invalid option -- 'p'
Usage: grep [OPTION]... PATTERNS [FILE]...
Try 'grep --help' for more information.
```

If the underlying tool has an option that looks like an unknown option argument to `west sign`, you similarly need `--`. There's nothing west specific going on here that I can tell, nor do I see an error that needs correcting.

Am I missing something?

```
it would still be nice if Zephyr's instructions were more directly helpful.
```

Beyond the documentation I pasted, this particular example is right there in the `west help sign` output:

```
imgtool
-------

Currently, MCUboot's 'imgtool' tool is supported. To build a signed
binary you can load with MCUboot using imgtool, run this from your
build directory:

   west sign -t imgtool -- --key YOUR_SIGNING_KEY.pem
```

So again I'm not seeing what's missing.

pabigot, 2020-04-07:
OK, I understand your position on `--`'.

For the "directly helpful": yes, `--key` is shown.  I needed to figure out how to encode a version number that showed up in the dfu-util image list, and it wasn't clear to me whether that was something Zephyr put into the image somehow or that was added by imgtool, and there were no clues.  Hopefully next time I try to get somewhere on dfu I'll remember all this.  Thanks for being patient.

mbolivar-nordic, 2020-04-08:
> I needed to figure out how to encode a version number that showed up in the dfu-util image list, and it wasn't clear to me whether that was something Zephyr put into the image somehow or that was added by imgtool, and there were no clues.

That is also in the `west help sign` output:

```
A default version number of 0.0.0+0 is used (which can be overridden by
passing "--version x.y.z+w" after "--key")
```