zephyrproject-rtos/zephyr #24108: https GET request is failed for big file download.

Navin-Sankar, 2020-04-05:
The following PR #24107 adds the hawkbit FOTA libary support to update embedded device using http and https mode. HTTP mode is working fine(docker image). Hawkbit https server is build with self signed certificate to run in local network. The HTTPS mode is working when the response size is < 2Kb. While downloading the update image from the server, download is failed and client sent the encrypted alert message to the server to stop the communication. 

Wireshark file is attached.

server.ip = 192.168.1.108 & client.ip = 192.168.1.109
[frdm_https.zip](https://github.com/zephyrproject-rtos/zephyr/files/4434465/frdm_https.zip)

jukkar, 2020-04-08:
Could you try to increase the value of `CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN` and retry. That option determines what is the maximum size of the encrypted packet, default value is 1500 bytes.

Navin-Sankar, 2020-04-14:
Increased the size of **CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN** to **4096**. But still getting the same response. Attached tls-config file
[image]
**overlay-tls.conf**
````
CONFIG_MAIN_STACK_SIZE=8192

# TLS configuration
CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_BUILTIN=y
CONFIG_MBEDTLS_ENABLE_HEAP=y
CONFIG_MBEDTLS_HEAP_SIZE=30000
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=4096

CONFIG_NET_SOCKETS_SOCKOPT_TLS=y
````

jukkar, 2020-04-30:
I do not have hawkbit environment setup. Do you have any other way to replicate this issue?

Navin-Sankar, 2020-05-05:
Changed the TLS Kconfig as per the sample big_http_download.

From 
````
CONFIG_MBEDTLS_HEAP_SIZE=30000
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=4096
````
to 

````
CONFIG_MBEDTLS_HEAP_SIZE=60000
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=16384
````
now it's working. Thanks @jukkar

closed 2020-05-05 by jukkar