zephyrproject-rtos/zephyr #24119: STM32: SPI: Extend power saving SPI pin config to all stm32 series

erwango, 2020-04-06:
**Is your enhancement proposal related to a problem? Please describe.**
https://github.com/zephyrproject-rtos/zephyr/pull/24097 introdcued power saving compliant SPI pin configurations.
This is likely this configuration will be useful for other series as well, specially LP ones (stm32wb included)

**Describe the solution you'd like**
Extend #24097 to all stm32 series.

FRASTM, 2020-05-06:
For the stm32wb soc series, see https://github.com/zephyrproject-rtos/zephyr/issues/24119

FRASTM, 2021-03-29:
Like the power management control of the uart , there is a possible **spi_stm32_pm_control** in PR https://github.com/zephyrproject-rtos/zephyr/pull/32544

erwango, 2021-03-29:
@FRASTM request in this ticket is not the dynamic pin configuration handling in SPI driver (like proposed in #32544), but the static configuration, that would be effective even with `CONFIG_PM=n`. Please check what was done in #25033

aurel32, 2021-04-25:
With the new device tree bindings for SPI pins configuration, the fix can be done on all the STM32 series at the same time. This should be fixed by: https://github.com/zephyrproject-rtos/hal_stm32/pull/95

erwango, 2021-06-03:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/34786

closed 2021-06-03 by erwango