zephyrproject-rtos/zephyr #24133: Question: Context save/restore after deep sleep using device driver

tektronix-94, 2020-04-06:
Hello,

I'm having some issues related to power management and context restore after sleep using BlueNRG SOC with Cortex M0.

According to documentation context should be restored in `_sys_resume_from_deep_sleep()` function but this function is not supported on Cortex M0.
So, I tried to add device driver context restore in the Reset Vector handler. Unfortunately it does not work properly.

After wake up from deep sleep the application is resumed from the point where the system entered the deep sleep state. However, all the threads are stopped and k_sleep() does not work.

Where `_sys_resume_from_deep_sleep()` should be called for context restore on Cortex M0 architecture ?

How to save/restore stack of threads?

\

tektronix-94, 2020-04-06:
area: Power Management

pabigot, 2020-04-07:
I wouldn't expect the application to restart from deep sleep at the point where it called it.  It's generally expected to be equivalent to a system reset, potentially with some flags set somewhere system-/application-specific to indicate that a wake has occurred.

AFAIK there is no in-tree support for recovering state from deep sleep on any platform, though there's a non-functional stub that allows applications to link to it.  Only ARC and X86 seem to have boot-time paths that might cause it to be invoked automatically.

carlescufi, 2020-04-21:
@tektronix-94 I think the question is answered, though the functionality you want is not implemented yet. Could you close this if you agree?

tektronix-94, 2020-04-22:
> 
> 
> @tektronix-94 I think the question is answered, though the functionality you want is not implemented yet. Could you close this if you agree?

Sure.

closed 2020-06-25 by MaureenHelm