zephyrproject-rtos/zephyr #24142: NRF5340 PA/LNA doesnæt work

binkboyd, 2020-04-07:
**Describe the bug**
When enabling PA/LNA pin control the build fails with what appears to be both GPIO and PPID related errors. It doesn't look like the NRF53 PA/LNA code is implemented yet.

**To Reproduce**
Steps to reproduce the behavior:
1) 
add following to hci_rpmsg sample prj.conf, using the 
CONFIG_BT_CTLR=y
CONFIG_BT_CTLR_GPIO_PA=y
CONFIG_BT_CTLR_GPIO_PA_PIN=29
CONFIG_BT_CTLR_GPIO_LNA=y
CONFIG_BT_CTLR_GPIO_LNA_PIN=24

2)
west build samples/bluetooth/hci_rpmsg -b nrf5340_dk_nrf5340_cpunet
3)
Fails with lots of error, but seems to be both GPIO and ppi related:
...error: 'NRF_GPIO' undeclared (first use in this function); did you mean 'NRF_GPIOTE'?
 #define NRF_GPIO_PA     NRF_GPIO

...warning: implicit declaration of function 'hal_enable_palna_ppi_config'; did you mean 'hal_trigger_aar_ppi_config'? [-Wimplicit-function-declaration]

...error: 'HAL_ENABLE_PALNA_PPI' undeclared 

**Expected behavior**
Expected the sample to build and PA/LNA control to work.

**Impact**
My board is not useable as I have an external amplifier which cannot be controlled by the NRF5340.

**Environment (please complete the following information):**
- Latest NCS, so not up to date zephyr - but the affected file (radio.c) is the same as in latest zephyr master (7ba5e012f7f2e380b360df84f3a345fdec9cc234)

carlescufi, 2020-04-08:
@anangl would you please take a look at this? It's a matter of porting the PPI to DPPI

ioannisg, 2020-05-11:
Is this really a bug @carlescufi ? PA/LNA support has not been ported yet to nRF5340, so AFAIU, it has never worked, so the ticket should be labeled as feature/enhancement

carlescufi, 2020-05-26:
Yep, agreed @ioannisg. Let's relabel this as hardware support

SloMusti, 2020-11-24:
I have the same issue and would like to use this functionality.

cvinayak, 2020-12-22:
@binkboyd and @SloMusti 

Please try using #30950  and leave your comments here or in the PR.

Add the following to your application core MCU program, to permit the GPIO to be controlled by network core MCU.
```
	NRF_P0->PIN_CNF[29] =
		GPIO_PIN_CNF_MCUSEL_NetworkMCU << GPIO_PIN_CNF_MCUSEL_Pos;
	NRF_P0->PIN_CNF[24] =
		GPIO_PIN_CNF_MCUSEL_NetworkMCU << GPIO_PIN_CNF_MCUSEL_Pos;
```

PR #31336 (2021-04-21): Bluetooth: controller: nRF21540 FEM support for nRF52x and nRF53x Series
Added nRF21540 FEM support for nRF52x and nRF53x Series.

Thanks to @bjornspock for the FEM contributions and @auroraslb for the DPPI refactoring.

[image]

Fixes #24142.

~Contains cherry-picked commit from @auroraslb PR: #30732~
commit 35122d4b (2021-04-21): Bluetooth: controller: GPIO PA/LNA support for nRF53x

Ported the GPIO PA/LNA support for nRF53x by using a DPPI
channel.

Fixes #24142.
closed 2021-04-21 by carlescufi