zephyrproject-rtos/zephyr #24144: deadlock potential in nrf_qspi_nor

pabigot, 2020-04-07:
As pointed out [here](https://github.com/zephyrproject-rtos/zephyr/pull/24128#issuecomment-610235114) by @ball-hayden the QSPI driver for Nordic persistently invokes an operation but blocks for release of a semaphore whether or not the operation failed.  At least some error codes may indicate that the operation was never initiated, which would produce a deadlock in that case, as demonstrated in #24122.

carlescufi, 2020-04-07:
@kamlaz could you provide a fix?

kamlaz, 2020-04-08:
> @kamlaz could you provide a fix?

Yes, I will handle this.

pabigot, 2020-04-19:
I lost a couple hours due to running into this when updating the USB mass storage to support mounting a littlefs file system, so there's now a fix.  Many things that use wrappers to the SPI flash API are likely to fail to supply an aligned memory buffer and will produce a deadlock that is very hard to diagnose, so I've increased this to medium priority.

PR #24488 (2020-04-20): drivers: flash: nrfx_qspi_nor: fix semaphore errors
The code generally invoked qspi_wait_for_completion() within a locked
region without verifying that an operation was successfully initiated.
This caused a deadlock whenever the operation failed, e.g. because the
data buffer was not 4-byte aligned.  Update that function to take the
result of the operation and either wait for completion or release the
lock, depending on its value.

Also uniformly use the correct type for Nordic HAL error values, and
refactor the erase module so that the correct values are displayed in
the diagnostic when someth [... cut]
closed 2020-04-20 by carlescufi