zephyrproject-rtos/zephyr #24145: File system shell example mount littleFS issue on nrf52840_pca10056

overheat, 2020-04-07:
**Describe the bug**
I follow the instruction of this link to try it on nrf52840_pca10056 board:
file:///home/aaron/repos/zephyrproject/zephyr/doc/_build/html/samples/subsys/shell/fs/README.html

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nrf52840_pca10056 samples/subsys/shell/fs
2. west flash
Then:
uart:~$ fs mount littlefs /lfs

**Screenshots or console output**
```
[00:00:19.039,581] <inf> littlefs: LittleFS version 2.1, disk version 2.0
[00:00:19.039,825] <inf> littlefs: FS at NRF_FLASH_DRV_NAME:0xf8000 is 8 0x1000-byte blocks with 512 cycle
[00:00:19.039,825] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:19.039,916] <err> littlefs: Corrupted dir pair at 0 1
[00:00:19.039,947] <wrn> littlefs: can't mount (LFS -84); formatting
[00:00:19.124,267] <err> os: ***** MPU FAULT *****
[00:00:19.124,267] <err> os:   Data Access Violation
[00:00:19.124,267] <err> os:   MMFAR Address: 0xf8000
[00:00:19.124,298] <err> os: r0/a1:  0x000f8000  r1/a2:  0x00000001  r2/a3:  0x80000000
[00:00:19.124,298] <err> os: r3/a4:  0x4001e000 r12/ip:  0x0000cc65 r14/lr:  0x000062fb
[00:00:19.124,328] <err> os:  xpsr:  0x81000000
[00:00:19.124,328] <err> os: Faulting instruction address (r15/pc): 0x00006bf8
[00:00:19.124,328] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:19.124,359] <err> os: Current thread: 0x20000450 (unknown)
[00:00:23.573,577] <err> os: Halting system
```

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain (Zephyr SDK, ...)
 - Commit SHA: 093fec62e7bddb6c1f6ad8a204d992759ef8d7ac

pabigot, 2020-04-07:
On that platform the storage partition is in on-SOC flash.  Writes to that are disallowed unless `CONFIG_MPU_ALLOW_FLASH_WRITE=y`.  If you don't already have an LFS file system on the partition, the mount will attempt to create one, which causes an access error.

There's an `mpu.conf` overlay in that directory, but nothing says the conditions under which it has to be used.  The CMakeFiles.txt will add it for several Nordic platforms, but only if you specify the board using its non-deprecated name:

```
west build -b nrf52840dk_nrf52840 samples/subs/shell/fs
west flash
```

works:

```
uart:~$ *** Booting Zephyr OS build zephyr-v2.2.0-1249-g08511a5645aa  ***

uart:~$ fs mount littlefs /lfs
[00:00:06.466,339] <inf> littlefs: LittleFS version 2.1, disk version 2.0
[00:00:06.466,552] <inf> littlefs: FS at NRF_FLASH_DRV_NAME:0xf8000 is 8 0x1000-
byte blocks with 512 cycle
[00:00:06.466,552] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:06.466,644] <err> littlefs: Corrupted dir pair at 0 1
[00:00:06.466,674] <wrn> littlefs: can't mount (LFS -84); formatting
[00:00:06.637,939] <inf> littlefs: /lfs mounted
uart:~$ fs cd /lfs
uart:~$ fs write test 1 2 3 4
uart:~$ fs read test
File size: 4
00000000  01 02 03 04                                      ....
uart:~$ fs ls
test
uart:~$ 
```

pabigot, 2020-04-07:
Closing as this is due to #24158; it's not a fault in shell or littlefs.

> BTW, do we have documentation about filesystem and shell zephyr used.

I don't believe there's much other than the sample documentation, and the help infrastructure within the shell itself.

closed 2020-04-07 by pabigot

overheat, 2020-04-08:
@pabigot 
Yes, it works. Thanks!

tejlmand, 2020-04-08:
@overheat the underlying issue is fixed here: https://github.com/zephyrproject-rtos/zephyr/pull/24205
It should of course be possible to still use deprecated names, although we of course advice people to use the new names.