zephyrproject-rtos/zephyr #24147: nrf5340 pdk: BOARDS_ENABLE_CPUNET does not allow proper NET MCU configuration

ioannisg, 2020-04-07:
**Describe the bug**
A recently introduced Kconfig option, BOARDS_ENABLE_CPUNET controls whether remoteproc_mgr_boot() is called from nrf5340 Application MCU booting process.

remoteproc_mgr_boot() should be ideally called only once, and by the image that resets the nrf5340 NETWORK MCU. So BOARDS_ENABLE_CPUNET should be set by the  image that cooperates with the NETWORK MCU. If this is a Non-Secure image, then parts of the configuration done in remoteproc_mgr_config() are never executed.

If, instead, we allow BOARDS_ENABLE_CPUNET to be enabled from both Secure and Non-Secure Application MCU images, then the RESET of NET MCU is occurring twice, which is very confusing.

We also do not deal with (extreme) corner cases when the image responsible to reset the NET MCU does not own the shared memory.

**Expected behavior**
We need to make sure remoteproc_mgr_boot does not contain dead code.
We should really re-organize the code in the respective module, to ensure that all parts of the code are executed by the intended images.

carlescufi, 2020-04-07:
@ioannisg is this really a bug? what priority should it have?

ioannisg, 2020-07-20:
confirmed, so I re-open

PR #27053 (2020-07-23): Nrf5340 fix board enable cpunet help text and usage
Clarifying and correcting some thing in the usage of config BOARD_ENABLE_CPUNET option.

Fixes #24147

Summarizing the usage of BOARD_ENABLE_CPUNET

- if building a firmware without TrustZone awarness (TRUSTED_EXECUTION_SECURE=n, TRUSTED_EXECUTION_NONSECURE=n), then all actions in remoteproc_mgr_config are needed
- if building a strictly Secure firmware (TRUSTED_EXECUTION_SECURE=y), only the GPIO assignment is needed
- if building a strictly Non-Secure firmware (TRUSTED_EXECUTION_NONSECURE=y), only the shared mem initialization and the RESET  [... cut]
closed 2020-07-23 by carlescufi