zephyrproject-rtos/zephyr #24158: gap in support for deprecated Nordic board names

pabigot, 2020-04-07:
Something seems to be failing with the deprecated Nordic board names.

`samples/subsys/shell/fs/CMakeLists.txt` looks at the board name to determine whether the `mpu.conf` overlay is required.  The file was updated to use the new board names, and the addition is not performed for the deprecated name.

```
tirzah[346]$ rm -rf build && west build -b nrf52840dk_nrf52840 samples/subsys/shell/fs >/dev/null && grep CONFIG_MPU_ALLOW build/zephyr/.config 
Including boilerplate (Zephyr base): /mnt/nordic/zp/zephyr/cmake/app/boilerplate.cmake
CONFIG_MPU_ALLOW_FLASH_WRITE=y

tirzah[347]$ rm -rf build && west build -b nrf52840_pca10056 samples/subsys/shell/fs >/dev/null && grep CONFIG_MPU_ALLOW build/zephyr/.config 
Including boilerplate (Zephyr base): /mnt/nordic/zp/zephyr/cmake/app/boilerplate.cmake
CMake Warning at /mnt/nordic/zp/zephyr/cmake/app/boilerplate.cmake:200 (message):
  Deprecated BOARD=nrf52840_pca10056 name specified, board automatically
  changed to: nrf52840dk_nrf52840.
Call Stack (most recent call first):
  /mnt/nordic/zp/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /mnt/nordic/zp/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:17 (find_package)

# CONFIG_MPU_ALLOW_FLASH_WRITE is not set

```

pabigot, 2020-04-07:
@carlescufi This is the underlying cause of #24145.

carlescufi, 2020-04-08:
@tejlmand can you take a look please?
@pabigot thanks for the analysis

tejlmand, 2020-04-08:
This happens because the sample CMakeLists.txt itself tests the board name BEFORE renaming is done.

Hence, it doesn't know anything about deprecated boards.
Secondly, it also happens before boilerplate is sourced, making things trickier.

Of course the easy fix is to just update to test both new and deprecated boards, but as this also affect aliases, it should be considered if we can do something else.
https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/subsys/shell/fs/CMakeLists.txt#L5-L8

tejlmand, 2020-04-08:
actually, the CMake code in the linked file will never work as intended, cause `APPLICATION_SOURCE_DIR` is a variable being defined in boilerplate, and hence not available in code at that time it is used.

PR #24205 (2020-04-09): cmake: fixed issue with invalid usage of APPLICATION_SOURCE_DIR
Fixes: #24158

The shell/fs sample was using APPLICATION_SOURCE_DIR before it has been
defined.

This has now been fixed to use the proper way of overlaying board
specific settings using boards folder.
commit 0e31656e (2020-04-09): cmake: fixed issue with invalid usage of APPLICATION_SOURCE_DIR

Fixes: #24158

The shell/fs sample was using APPLICATION_SOURCE_DIR before it has been
defined.

This has now been fixed to use the proper way of overlaying board
specific settings using boards folder.
closed 2020-04-09 by carlescufi