zephyrproject-rtos/zephyr #24176: Where can I read PDR (packet delivery ratio)? Or number of TX/ACK packets?

BOZHENG001, 2020-04-08:
Dear All

I am right now developing BLE application based on `central_hr` and `peripheral_hr` on nRF52840 DK with zephyr 2.0.

Basically, I am looking for an API in zephyr to read PDR (packet delivery ratio) from no matter master or slave devices, but obviously there is no API for this totally unimportant function based on BLE stack. 

Then I started to look into number of TX/ACK packets. Because we can always calculate PDR based on packets we sent and acknowledgments we received. I can find some clues from file `ull_conn.c`, but still I am not sure which functions inside are responsible for TX sent and ACK received.

Could anyone help me with that? All I need to do is actually adding a simple counter to the right places/functions for TX and ACK.

cvinayak, 2020-04-09:
Transmit is prepared here:
https://github.com/zephyrproject-rtos/zephyr/blob/4547bfb452b5d33b9ba052890ec64afa38c51d53/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_master.c#L137
, and here:
https://github.com/zephyrproject-rtos/zephyr/blob/4547bfb452b5d33b9ba052890ec64afa38c51d53/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_conn.c#L214

Do not count your transmit when in this control path, here:
https://github.com/zephyrproject-rtos/zephyr/blob/4547bfb452b5d33b9ba052890ec64afa38c51d53/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_master.c#L205
, and here:
https://github.com/zephyrproject-rtos/zephyr/blob/4547bfb452b5d33b9ba052890ec64afa38c51d53/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_conn.c#L242

Acknowledgement is determined here:
https://github.com/zephyrproject-rtos/zephyr/blob/4547bfb452b5d33b9ba052890ec64afa38c51d53/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_conn.c#L665

Please close the question, if you are satisfied.

BOZHENG001, 2020-04-09:
@cvinayak Thanks for your answer! Totally solved my question.

closed 2020-04-09 by BOZHENG001