zephyrproject-rtos/zephyr #24178: CI: extra_args from sanitycheck *.yaml do not propogate to cmake

finikorg, 2020-04-08:
extra_args from *.yaml do not go to the build system.

**Reproduce**
extra_args: "-DDEBUG_PRINTF=1" from samples/philosophers/sample.yaml is lost.
```
$ sanitycheck -p native_posix -T zephyr/samples/philosophers/
```

Output:
```
...
CMake Warning:
  Manually-specified variables were not used by the project:

    -DDEBUG_PRINTF
...
```

and

```
DEBUG   - Calling cmake with arguments: ['-B/home/niko/work/zephyrproject/sanity-out/native_posix/samples/philosophers/sample.kernel.philosopher', '-S/home/niko/work/zephyrproject/zephyr/samples/philosophers', '-DEXTRA_CFLAGS="-Werror ', '-DEXTRA_AFLAGS=-Wa,--fatal-warnings', '-DEXTRA_LDFLAGS="-Wl,--fatal-warnings', '-GUnix Makefiles', '-D-DDEBUG_PRINTF=1', '-DBOARD=native_posix']
```

mbolivar-nordic, 2020-04-08:
I don't think I'm the right assignee.

nashif, 2020-04-11:
so, this is not an issue in sanitycheck or west.. The fact that cmake reports about the variable means it was passed correctly to the sample.
Looks like cmake has no idea what to with this and some additional handling is needed in the cmake file, fixed for this sample. If we do not want this to go through cmake then i think we should add

```
-DEXTRA_CFLAGS="-DDEBUG_PRINTF=1"
```
instead, not sure yet.

closed 2020-04-12 by nashif

finikorg, 2020-04-14:
We can keep this bug to handle also other parameters in extra_args.

PR #24288 (2020-04-12): samples: philosophers: evaluate variables coming from command line
The variables passed by sanitycheck and west were being ignored because
cmake knows nothing about them.

Fixes #24178
commit acc0e902 (2020-04-12): samples: philosophers: evaluate variables coming from command line

The variables passed by sanitycheck and west were being ignored because
cmake knows nothing about them.

Fixes #24178
PR #24330 (2020-04-14): samples: philosophers: Fix extra_args not used
Fixed extra_args parameter DEBUG_PRINTF in sample.yaml not used.

Fixes #24178