zephyrproject-rtos/zephyr #24181: Snprintk used at many place while dummy build if CONFIG_PRINTK is undef

XavierChapron, 2020-04-08:
**Describe the bug**
`snprintk()` defined in `include/sys/printk.h` has a dummy version in case `CONFIG_PRINTK` is not defined:
```c
static inline __printf_like(3, 4) int snprintk(char *str, size_t size,
					       const char *fmt, ...)
{
	ARG_UNUSED(str);
	ARG_UNUSED(size);
	ARG_UNUSED(fmt);
	return 0;
}
```

However, it is used in many places (`git grep snprintk | wc` gives 213 matchs), including some that must not be "optimized" when logs are removed:
- it is used to create AT commands in ublox-sara-r4, wncm14a2a, esp drivers
- it is used to create JSON payload in lib/os/json.c
- it is used in many Bluetooth related functions.

**To Reproduce**
Build any project without `CONFIG_PRINTK`.
To be sure, that it's not re-enable by (`CONFIG_BOOT_BANNER` for example) also remove `CONFIG_CONSOLE` and `CONFIG_LOG`.
See that `snprintk()` as been "optimized" in generated `.map` file.

**Expected behavior**
`snprintk()` should either be:
- Always built with a full implementation.
- Not exposed in `printk.h` and only used in functions already optimized when `CONFIG_PRINTK` is not defined.

**Impact**
This become really dangerous if `CONFIG_PRINTK` is not defined.

**Environment :**
 - On Zephyr master at the time of writtting this (SHA: 5b991aa09a7902f4ac3a9f506c46e8fa8403001f)

XavierChapron, 2020-04-08:
Currently both `snprintf()` and `snprintk()` are used.
This leads to having both function in generated binary which is an unnecessary duplicate increasing flash footprint.

Should this be in this issue or in another one?

jukkar, 2020-04-08:
> Should this be in this issue or in another one?

Sounds like a separate issue to me.

IMHO best option would be to always have a full implementation of snprintk().

XavierChapron, 2020-04-08:
> > Should this be in this issue or in another one?
> 
> Sounds like a separate issue to me.
> 
> IMHO best option would be to always have a full implementation of snprintk().

Thank for your reply.
I have created a separated issue.

carlescufi, 2020-04-14:
@nashif found this commit which should've decoupled CONFIG_PRINTK and snprintk. https://github.com/zephyrproject-rtos/zephyr/commit/d76ae46c0c578b2cddbd2d9cf9b9d910af8e2c40

andrewboie, 2020-04-14:
wait a second...snprintk does not depend on CONFIG_PRINTK

what's the actual problem here?

```
static inline __printf_like(3, 4) int snprintk(char *str, size_t size,
					       const char *fmt, ...)
{
	ARG_UNUSED(str);
	ARG_UNUSED(size);
	ARG_UNUSED(fmt);
	return 0;
}
```

I just checked and this code doesn't exist.

Are you on some old version of Zephyr?

closed 2020-04-14 by andrewboie

XavierChapron, 2020-04-15:
@andrewboie You are correct, I've not correctly checked zephyr last version and therefore I've not see your commit that is fixing this issue. I'm sorry for the annoyance.