zephyrproject-rtos/zephyr #24193: Issue with launching examples on custom board (after succesfull build)

kubazdz, 2020-04-08:
**Describe the bug**
I have a custom board with nrf52832. I made whole board files set and try to launch blinky example on it with additional uart printk (board has uart). Example builds successfully (with custom board specified) but after flashing there is no uart output (even "booting Zephyr OS...") and the led doesn't blink. When i flash the same example (without rebuilding) to nrf52_pca10040 it works perfectly. To werify hardware and pinout I did the same example with nrf sdk16 and it worked on both custom and pca boards so the hardware is fine.

**Expected behavior**
Working blinky example on both boards

**Impact**
Can't work on a simple custom board

**Screenshots or console output**
My files:

Kconfig.board:
```
# xxx_nrf52832 board configuration

# Copyright (c) 2016 Nordic Semiconductor ASA
# SPDX-License-Identifier: Apache-2.0

config BOARD_XXX_NRF52832
	bool "xxx_nrf52832"
	depends on SOC_NRF52832_QFAA
```

Kconfig.defconfig:
```
# xxx_nrf52832 board configuration

# Copyright (c) 2016 Nordic Semiconductor ASA
# SPDX-License-Identifier: Apache-2.0

if BOARD_XXX_NRF52832

config BOARD
	default "xxx_nrf52832"

endif # BOARD_XXX_NRF52832
```

xxx_nrf52832.yaml:
```identifier: xxx_nrf52832
name: yyy xxx_nrf52832
type: mcu
arch: arm
toolchain:
  - zephyr
  - gnuarmemb
  - xtools
ram: 64
flash: 512
supported:
  - gpio
  - uart
  - watchdog
```

xxx_nrf52832_defconfig:
```# SPDX-License-Identifier: Apache-2.0

CONFIG_ARM=y
CONFIG_SOC_FAMILY_NRF=y
CONFIG_SOC_SERIES_NRF52X=y
CONFIG_SOC_NRF52832_QFAA=y
CONFIG_BOARD_XXX_NRF52832=y

# Enable MPU
CONFIG_ARM_MPU=y

# enable GPIO
CONFIG_GPIO=y

# enable uart driver
CONFIG_SERIAL=y
CONFIG_UART_0_NRF_UART=y

# enable console
CONFIG_CONSOLE=y
CONFIG_UART_CONSOLE=y
```

xxx_nrf52832.dts:
```/*
 * Copyright (c) 2017 Shawn Nock <shawn@monadnock.ca>
 * Copyright (c) 2017 Linaro Limited
 *
 * SPDX-License-Identifier: Apache-2.0
 */

/dts-v1/;
#include <nordic/nrf52832_qfaa.dtsi>

/ {
	model = "yyy xxx_nrf52832";
	compatible = "yyy,xxx_nrf52832","nordic,nrf52832-qfaa",
			"nordic,nrf52832";

	chosen {
		zephyr,console = &uart0;
		zephyr,shell-uart = &uart0;
		zephyr,bt-mon-uart = &uart0;
		zephyr,bt-c2h-uart = &uart0;
		zephyr,sram = &sram0;
		zephyr,flash = &flash0;
		zephyr,code-partition = &slot0_partition;
	};

	leds {
		compatible = "gpio-leds";
		led0: led_0 {
			gpios = <&gpio0 18 GPIO_INT_ACTIVE_LOW>;
			label = "Green LED 0";
		};
	};

	/* These aliases are provided for compatibility with samples */
	aliases {
		led0 = &led0;
	};
};

&gpiote {
	status = "okay";
};

&gpio0 {
	status = "okay";
};

&uart0 {
	status = "okay";
	compatible = "nordic,nrf-uart";
	current-speed = <115200>;
	tx-pin = <6>;
	rx-pin = <8>;
	rts-pin = <5>;
	cts-pin = <7>;
};

&flash0 {
	/*
	 * For more information, see:
	 * http://docs.zephyrproject.org/latest/guides/dts/index.html#flash-partitions
	 */
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x00000000 0xc000>;
		};
		slot0_partition: partition@c000 {
			label = "image-0";
			reg = <0x0000C000 0x32000>;
		};
		slot1_partition: partition@3e000 {
			label = "image-1";
			reg = <0x0003E000 0x32000>;
		};
		scratch_partition: partition@70000 {
			label = "image-scratch";
			reg = <0x00070000 0xa000>;
		};
		storage_partition: partition@7a000 {
			label = "storage";
			reg = <0x0007a000 0x00006000>;
		};
	};
};
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain Zephyr SDK
 - Zephyr version 2.1.0

mbolivar-nordic, 2020-04-08:
Can you attach a debugger and see if control flow reaches main()?

kubazdz, 2020-04-09:
@mbolivar-nordic actually I did upgrade to latest (2.2.99) version and it works now

carlescufi, 2020-04-09:
Closing since this is fixed now.

closed 2020-04-09 by carlescufi