zephyrproject-rtos/zephyr #24194: Unknown message received by the node

rajputh54, 2020-04-08:
**Describe the bug**
I have setup for two devices Node-A (nRF52840 pca10056) and Node-B (nRF52 pca10040).
Node-B sends report continuously at  interval of 2550 msec interval to Node-A.
Node-A receives a message with expected value but at the same time it gets one more message with garbage values. 
Message length is 5 bytes so it's unsegmented message.

**To Reproduce**
Steps to reproduce the behavior:
1. Follow zephyr setup [guide](https://docs.zephyrproject.org/latest/getting_started/index.html)
1. $ git clone https://github.com/rajputh54/reproduce-zephyr-issues.git
2. $ cd reproduce-zephyr-issues
3. Copy node-a and node-b folders in zephyrproject or zephyr directory 
4. Compile node-a and node-b for nRF52 boards and flash the same.
6. Now install nRF mesh app in Android phone.
7. Provison node A and B
8. configure Generic On Off Client (element section)
     - For Node-A, publish to 0xC000 and subscribe to 0xC001
     - For Node B publish to 0xC001 and subscribe to 0xC000
9. Now press button-1 on Node-A device. Node-B will starts sending dummy envi report to Node-A.
you can see in uart terminal or RTT viewer

**Expected behavior**
Node-A should get only one call to function envi_report every 2550 msec and print logs
`temperature:32 airp:18 humidity:84`
but Node-A gets the envi_report function call twice. First time it receives correct value  and second call it gives some garbage value for airp and humidity
`temperature:32 airp:18 humidity:84`  Expected values
`temperature:32 airp:3602 humidity:260` Garbage value for airp and humidity.

**Impact**
annoyance

**Screenshots or Console output**
Node-B (sender) console output
[image]

Node-A (receiver) console output 
[image]

**code snippet**
To get full code please have a look at node-a/src/main.c and node-b/src/main.c 
\`\`\`
////Node-A envi_report function
static void envi_report(struct bt_mesh_model *model,
                struct bt_mesh_msg_ctx *ctx,
                struct net_buf_simple *buf)
{
        s8_t temperature = net_buf_simple_pull_u8(buf);
        u16_t airp = net_buf_simple_pull_le16(buf);
        u16_t humidity = net_buf_simple_pull_le16(buf);

       printk("------------------------\n");
       printk("-->[0x%04x] ", ctx->addr);
       printk("temperature:%hhd airp:%hu humidity:%hu\n", temperature, airp, humidity);
       printk("-------------------------\n");
}
\`\`\`

\`\`\`
//////Node-B envi report function
static inline void snd_envi_report(void)
{
        int ret;
        u16_t airp = 18, humidity = 84;
        s8_t temperature = die_temperature();
        #if 0
        printk("[<--0x%04x]:timestamp = %u\n", cfg->node_addr, timestamp);
        #endif
        bt_mesh_model_msg_init(mod_pub->msg, OP_ENVI_REPORT);
        net_buf_simple_add_u8(mod_pub->msg, temperature);
        net_buf_simple_add_le16(mod_pub->msg, airp);
        net_buf_simple_add_le16(mod_pub->msg, humidity);
        ret = bt_mesh_model_publish(mod_cli);
        if (ret) {
                printk("\n%s:bt_mesh_model_publish failed, ERR = %d\n", __func__, ret);
                return;
        }
        printk("------------------------\n");
        printk("temperature:%hhd airp:%hu humidity:%hu\n", temperature, airp, humidity);
        printk("-------------------------\n");
}
K_THREAD_DEFINE(envi_tid, 2048, envi_thread, NULL, NULL, NULL, 7, 0, K_NO_WAIT);
static void envi_thread(void)
{
        while (1) {

                if (envi_report_flg) {
                        k_sleep(K_MSEC(2500));
                        snd_envi_report();
                } else {
                        k_thread_suspend(envi_tid);
                }
        }
}
\`\`\`

**Environment**
 - OS:  Linux
 - Toolchain Zephyr SDK
 - Commit: b74cad5a5d0f35905c4ab3d719bdc6b2794bc940 (one can try with latest commit)

LingaoMeng, 2020-04-09:
@rajpadm  Can you send different data for every publish message ?
I'm notice that number `18` eq `0b0001 0010` and number `3602` eq `0b1110 0001 0010`

LingaoMeng, 2020-04-09:
While I'm found some buffer issure https://github.com/rajputh54/reproduce-zephyr-issues/blob/eb2f049467e778d20885b20561a9d6b48753b163/node-b/src/main.c#L118-L125

As you say above, every publish will occupy `five` octers (exclude opcode), but you allocated only `four` octes.

rajputh54, 2020-04-09:
Thanks a lot for your help !!!
As you suggested, It's working fine after adding enough pub buffer,

closed 2020-04-09 by rajputh54