zephyrproject-rtos/zephyr #24197: Reduce snprintf and snprintk footprint

XavierChapron, 2020-04-08:
**Is your enhancement proposal related to a problem? Please describe.**
Both `snprintk()` and `snprintf()` are used and therefore linked in the app binary.
This is waste of flash to have both when they basically have mostly the same behavior.

**Describe the solution you'd like**
To me, the best solution would be:
- Add a `Kconfig` variable to either use `snprintf()` or `snprintk()` based on the user and libraries needs.

**Describe alternatives you've considered**
Another safer and easier solution could be:
- Always use `snprintf()` as `snprintk()` is a lightweight implementation of `snprintk()`.

XavierChapron, 2020-04-08:
This issue is linked to https://github.com/zephyrproject-rtos/zephyr/issues/24181.

carlescufi, 2020-04-21:
@XavierChapron I assume that this is still an issue even though we closed #24181?

andrewboie, 2020-04-21:
snprintk() is part of the Zephyr kernel. It is small, does not implement the entire specification, does not use much stack space.

snprintf() is part of the C library. It is variously implemented in newlib, minimal libc, etc. It is standards compliant but much larger than snprintk() and uses a lot more stack space.

> Always use snprintf() as snprintk() is a lightweight implementation of snprintk().

This doesn't quite make sense to me can you rephrase?

> Both snprintk() and snprintf() are used and therefore linked in the app binary.

> Add a Kconfig variable to either use snprintf() or snprintk() based on the user and libraries needs.

I don't think this will work because snprintk() is not directly equivalent to snprintf()

XavierChapron, 2020-04-22:
> snprintk() is part of the Zephyr kernel. It is small, does not implement the entire specification, does not use much stack space.

I agree with that, it is clearly small, and that is great. If we add everything it is weighing around 1kB in my binary, so definitively a small implementation.

```
 .text._printk_dec_ulong
                0x000000000000e02c       0x98 libzephyr.a(printk.c.obj)
 .text.__printk_hook_install
                0x000000000000e0c4        0xc libzephyr.a(printk.c.obj)
                0x000000000000e0c4                __printk_hook_install
 .text.vsnprintk
                0x000000000000e0d0       0x28 libzephyr.a(printk.c.obj)
                0x000000000000e0d0                vsnprintk
 .text.arch_printk_char_out
                0x0000000000030930        0x4 libzephyr.a(printk.c.obj)
                0x0000000000030930                arch_printk_char_out
 .text.print_err
                0x0000000000030934       0x1c libzephyr.a(printk.c.obj)
 .text.str_out  0x0000000000030950       0x26 libzephyr.a(printk.c.obj)
 .text.z_vprintk
                0x0000000000030976      0x2e0 libzephyr.a(printk.c.obj)
                0x0000000000030976                z_vprintk
 .text.printk   0x0000000000030c56       0x1a libzephyr.a(printk.c.obj)
                0x0000000000030c56                printk
 .text.snprintk
                0x0000000000030c70       0x1a libzephyr.a(printk.c.obj)
                0x0000000000030c70                snprintk
```

> snprintf() is part of the C library. It is variously implemented in newlib, minimal libc, etc. It is standards compliant but much larger than snprintk() and uses a lot more stack space.

I have not looked at the newlib or minimal lib implementation, but they are indeed heavier. For newlib, using `CONFIG_NEWLIB_LIBC_FLOAT_PRINTF=n` when not necessary is already saving lot and lot of flash.

My point is that they are very similar, for every supported input of snprintk, snprintf will give the same input.
This means that (excluding the fact that snprintf use probably more stack) using something like `#define z_vprintk z_vprintf` instead of the full implementation of snprintk should work.
But this will only spare ~1kB of flash.

However, in many situations, snprintk is definitively enough. When looking at snprintf usage in zephyr, I don't see many situations where snprintk will not have done the exact same job.
If every not necessary usage of snprintk is removed, then it will be link only when really necessary.

@carlescufi @andrewboie What are your though about this.
In my opinion this should help decide if a driver should use snprintk or snprintf, the rule could be clear:
- If snprintk is enough, go for it
- However, if you really need snprintf, then you have no other choice.

stephanosio, 2020-04-22:
If you really would like to achieve what you suggest, I think it would make more sense to add an option that globally does:
```
#define snprintf(s_, sz_, f_, ...) snprintk((s_), (sz_), (f_), __VA_ARGS__)
```

**Implementations for reference**
https://github.com/zephyrproject-rtos/zephyr/blob/1d9b08aef5cd814406ae068f602cab54fdf2805f/lib/os/printk.c#L534

https://github.com/zephyrproject-rtos/zephyr/blob/ed139948a54bdb13bdafed57b1f00b68f0610890/lib/libc/minimal/source/stdout/sprintf.c#L30-L31

XavierChapron, 2020-04-22:
@stephanosio Why do you think that this as more sense ?
As you mention, some "advanced" functions will not work.
Whereas if all snprintf call that are not "advanced" are changed to snprintk, there is manual job but at the end no risk:
- If an snprintf remains in the code linked, then snprintf will be embedded in the binary.
- if no snprintf remains, then it should not be embedded.

carlescufi, 2020-05-06:
@XavierChapron could you try to reformulate your proposal either in words or in a draft Pull Request so we can try to move it forward?

XavierChapron, 2020-05-07:
@carlescufi I've drafted a PR so we can discuss about potential solutions: https://github.com/zephyrproject-rtos/zephyr/pull/25069

carlescufi, 2020-05-26:
Closing this since #25069 took care of it

closed 2020-05-26 by carlescufi