zephyrproject-rtos/zephyr #24200: USB GET_INTERFACE response always 0, even when an alternate setting is used

AndersAtEndian, 2020-04-08:
**Describe the bug**

The response to a USB GET_INTERFACE is always 0, see the following excerpt from subsys/usb/usb_device.c. 

```
	case REQ_GET_INTERFACE:
		/* there is only one interface, return n-1 (= 0) */
		data[0] = 0U;
		*len = 1;
		break;
```
But according to USB Specification Revision 2.0 (usb_20.pdf from usb.org) section 9.4.4 GET_INTERFACE should return "selected alternate setting for the specified interface."

**To Reproduce**

There is no (or at least I can't find one that builds for my board) sample that has multiple USB alternate settings; but it is possible to reproduce using a patch to subsys/usb/class/loopback.c and samples/subsys/usb/testusb. 

1. Apply the following patch.

```
diff --git a/subsys/usb/class/loopback.c b/subsys/usb/class/loopback.c
index 754c2a4906..df35aee390 100644
--- a/subsys/usb/class/loopback.c
+++ b/subsys/usb/class/loopback.c
@@ -30,6 +30,7 @@ struct usb_loopback_config {
        struct usb_if_descriptor if0;
        struct usb_ep_descriptor if0_out_ep;
        struct usb_ep_descriptor if0_in_ep;
+       struct usb_if_descriptor if0_1;
 } __packed;

 USBD_CLASS_DESCR_DEFINE(primary, 0) struct usb_loopback_config loopback_cfg = {
@@ -65,6 +66,19 @@ USBD_CLASS_DESCR_DEFINE(primary, 0) struct usb_loopback_config loopback_cfg = {
                .wMaxPacketSize = sys_cpu_to_le16(CONFIG_LOOPBACK_BULK_EP_MPS),
                .bInterval = 0x00,
        },
+
+       .if0_1 = {
+               .bLength = sizeof(struct usb_if_descriptor),
+               .bDescriptorType = USB_INTERFACE_DESC,
+               .bInterfaceNumber = 0,
+               .bAlternateSetting = 1,
+               .bNumEndpoints = 0,
+               .bInterfaceClass = CUSTOM_CLASS,
+               .bInterfaceSubClass = 0,
+               .bInterfaceProtocol = 0,
+               .iInterface = 0,
+       },
+
 };
 /* usb.rst config structure end */
```
2. Build the testusb sample (my example below for the NXP RT1020 EVK). 

```
west build -p auto -b mimxrt1020_evk samples/subsys/usb/testusb
```
3. Flash.
4. Run USB Chapter 9 TEsts (USB 2 Devices) in the compliance tool from usb.org (USB 3 Gen X).

The test will fail in Configured State - TD 9.4 with an error message "A successful GetInterface request must return the alternate setting set by a prior call to SetInterface". 

**Expected behavior**

Compliance tests should pass.

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Toolchain: Zephyr SDK 0.11.2
 - Zephyr master b74cad5a5d0f35905c4ab3d719bdc6b2794bc940.

emob-nordic, 2020-04-09:
First of all, whats the use case to add alternate setting for such device?
In order to make it work i suppose that [custom_handler](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/class/loopback.c#L169) must be implemented but I might be wrong as I took only i brief look into this. I cannot say when I will have time to do it as I am a bit busy right now and I do not see clear use case for that right now.
The current code is indeed incorrect but for all the classes and functionalities we support just right now its sufficient. 
Alternate settings are commonly used in audio class. Implementation could be found here #21526.
I think it would be worthwhile to implement it in audio for sure and as an author I will keep that in mind as for now it is missing.

AndersAtEndian, 2020-04-09:
Sorry for being a bit unclear - the example was just the quickest way I could find to create a patch that would fail in the USB compliance test. I'm tinkering with audio stuff and it was in that context that the issue was found. 

I can create a PR for a more generic solution; store the alternate setting after a successful USB SET_INTERFACE and returning that in the GET_INTERFACE request.

emob-nordic, 2020-04-09:
Ah, okay now I get this. If You want to post a PR it would be great. One question as You said 'generic solution'. Where do You gonna store the info about alternate setting?

AndersAtEndian, 2020-04-09:
Good question. If zephyr only supports one interface would it be out of place to store it in struct usb_dev_priv?

emob-nordic, 2020-04-09:
Until now it supported no alternate settings but many interfaces. For example if You configure 2x cdc there will be 4 interfaces in total. 

It cant be in usb_dev_priv as on this level You are not possible to guess the total number of interfaces so there is no possibility to create a table or whatsoever to store this information.
I think the only way now is to implement [custom_handler](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/usb/usb_device.h#L161) as it is called prio the [std_interface_req()](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/usb_device.c#L914), but this will get tricky as well because of how the set_interface is handled..

AndersAtEndian, 2020-04-09:
Thanks! I'll look into that.

emob-nordic, 2020-04-09:
We can follow up on Slack, feel free to ping me there.

jfischer-phytec-iot, 2020-04-09:
> There is no (or at least I can't find one that builds for my board) sample that has multiple USB alternate settings;

CDC ECM has alternate settings for the second interface:
[CDC ECM Interface 1 | bAlternateSetting = 0](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/class/netusb/function_ecm.c#L122) and [CDC ECM Interface 1 | bAlternateSetting = 1](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/class/netusb/function_ecm.c#L136)

emob-nordic, 2020-08-10:
This issue will be addressed during usb device stack rework and shall not remain stalled.

PR #36149 (2021-06-17): usb: store alternate interface settings
Fixes request hijack through Audio class workaround in composite configuration.
Fixes: #24200
commit 1e71cef1 (2021-06-17): usb: store alternate interface settings

Store alternate interface settings and
return alternate on GetInterface request.

Fixes: #24200
closed 2021-06-17 by carlescufi