zephyrproject-rtos/zephyr #24207: fcb test fails on nRF52840-DK

mbolivar-nordic, 2020-04-08:
**Describe the bug**

I'm getting a test failure on tests/subsys/fs/fcb for nrf52840dk_nrf52840.

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyrproject/zephyr
2. west build -b nrf52840dk_nrf52840 -p -t flash -s tests/subsys/fs/fcb
3. test fails on device serial console

**Expected behavior**

The board is listed in platform_whitelist inside tests/subsys/fs/fcb/testcase.yaml, so I expected this to pass.

**Impact**

Unclear.

**Screenshots or console output**

```
Running test suite test_fcb
===================================================================
starting test - fcb_test_len
PASS - fcb_test_len
===================================================================
starting test - fcb_test_init
PASS - fcb_test_init
===================================================================
starting test - fcb_test_empty_walk
PASS - fcb_test_empty_walk
===================================================================
starting test - fcb_test_append

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/fs/fcb/src/fcb_test_append.c:30: fcb_test_append: (rc == 0 is false)
flash_area_write call failure
FAIL - fcb_test_append
===================================================================
starting test - fcb_test_append_too_big

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/fs/fcb/src/fcb_test_append_too_big.c:42: fcb_test_append_too_big: (rc == 0 is false)
fcb_append call failure
FAIL - fcb_test_append_too_big
===================================================================
starting test - fcb_test_append_fill
PASS - fcb_test_append_fill
===================================================================
starting test - fcb_test_rotate
PASS - fcb_test_rotate
===================================================================
starting test - fcb_test_multi_scratch
PASS - fcb_test_multi_scratch
===================================================================
starting test - fcb_test_last_of_n
PASS - fcb_test_last_of_n
===================================================================
starting test - fcb_test_reset

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/fs/fcb/src/fcb_test_reset.c:82: fcb_test_reset: (rc == 0 is false)
flash_area_write call failure
FAIL - fcb_test_reset
===================================================================
Test suite test_fcb failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment:**

 - OS: linux
 - Toolchain Zephyr SDK 0.11.1
 - Commit SHA or Version used: recent master (e.g. d35567b375)

mbolivar-nordic, 2020-04-08:
@nvlsianpu not sure if you are the right assignee. Please feel free to change.

carlescufi, 2020-04-14:
@nvlsianpu should we deprecate FCB at this point?

mbolivar-nordic, 2020-04-14:
> @nvlsianpu should we deprecate FCB at this point?

Ah, so this is the one which might get deprecated. I couldn't remember the status of all these storage backends.

nvlsianpu, 2020-04-15:
> @nvlsianpu should we deprecate FCB at this point?

Not today - we can deprecate fcb at some point, but it was not problematic for user and initially it was the default setting back-end, so better to make a query before.

PR #24375 (2020-04-17): tests/subsys/fs/fcb: Fix flash_area_write failure on NRF
The commit fixes problem with test attempting to perform 1 byte writes
to flash, which are not emulated by default.

Fixes #24207 issue.
closed 2020-04-17 by ioannisg