zephyrproject-rtos/zephyr #24215: Couldn't flash image into up_squared using misc.py script.

chen-png, 2020-04-09:
**Describe the bug**
for up_squared and acrn, we used the scripts/west_commands/runners/misc.py to flash a image, but now it met an flash error.
the related commit is 3b49d0159, i reverted my repo before this commit, it was ok.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b up_squared -p auto samples/hello_world
2. west flash up_squared.sh
3. See error

**Screenshots or console output**
Traceback (most recent call last):
  File "/home/ztest/.local/bin/west", line 11, in <module>
    sys.exit(main())
File "/home/ztest/.local/lib/python3.6/site-packages/west/commands.py", line 116, in run
    self.do_run(args, unknown)
  File "/home/ztest/work/zephyrproject/zephyr/scripts/west_commands/flash.py", line 30, in do_run
    do_run_common(self, my_args, runner_args)
  File "/home/ztest/work/zephyrproject/zephyr/scripts/west_commands/run_common.py", line 147, in do_run_common
    runner_config = load_runners_yaml(runners_yaml, user_args)
  File "/home/ztest/work/zephyrproject/zephyr/scripts/west_commands/run_common.py", line 285, in load_runners_yaml
    config = yaml.safe_load(f.read())
  File "/home/ztest/.local/lib/python3.6/site-packages/yaml/__init__.py", line 162, in safe_load
    return load(stream, SafeLoader)
  File "/home/ztest/.local/lib/python3.6/site-packages/yaml/__init__.py", line 114, in load
    return loader.get_single_data()
  File "/home/ztest/.local/lib/python3.6/site-packages/yaml/constructor.py", line 41, in get_single_data
    node = self.get_single_node()
  File "/home/ztest/.local/lib/python3.6/site-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/home/ztest/.local/lib/python3.6/site-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/home/ztest/.local/lib/python3.6/site-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/home/ztest/.local/lib/python3.6/site-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/home/ztest/.local/lib/python3.6/site-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  **File "/home/ztest/.local/lib/python3.6/site-packages/yaml/parser.py", line 439, in parse_block_mapping_key
    "expected <block end>, but found %r" % token.id, token.start_mark)
yaml.parser.ParserError: while parsing a block mapping
  in "<unicode string>", line 2, column 1:
    runners:
    ^
expected <block end>, but found '['
  in "<unicode string>", line 20, column 2:
     []
     ^**

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.11.2
 - Commit ID: c413cca28044

chen-png, 2020-04-09:
@hakonfam  Could you take a look of this? it's related to commit 3b49d015

chen-png, 2020-04-09:
this blocked QA test, so changed it's priority to medium.

hakonfam, 2020-04-09:
Please check https://github.com/zephyrproject-rtos/zephyr/pull/24218, it fixed the issue on my side.

finikorg, 2020-04-09:
I have the same issue with up_square_adsp board, PR #22724 is failing.

closed 2020-04-09 by carlescufi

hakonfam, 2020-04-10:
> I have the same issue with up_square_adsp board, PR #22724 is failing.

Did #24218 fix the issue for you?

PR #24218 (2020-04-09): cmake: fix bug in generated runners.yaml
If a runner had no args, the format of the generated runners.yaml
was invalid due to missing indentation.

This commit fixes this issue by adding the required indentation.

Fixes #24215