zephyrproject-rtos/zephyr #24221: Do not run cron workflow on forks

ABOSTM, 2020-04-09:
**Describe the bug**
Since introduction of github Action: 

> .github/workflows/doc-publish.yml

I receive daily email about this action which runs on my fork.

> [ABOSTM/zephyr] Run failed: Doc build for Release or Daily - master (093fec6)

And I can imagine it will be the same for all zephyr forks that will be aligned on master branch.

**Expected behavior**
Do not run github action on all forks

There is an open thread to discuss this problem
https://github.community/t5/GitHub-Actions/Do-not-run-cron-workflows-in-forks/td-p/46756#

ABOSTM, 2020-04-09:
^^@galak

galak, 2020-04-09:
@ABOSTM a PR with a change to the workflow files is welcome.

ABOSTM, 2020-04-09:
No, I am not able to do so: I never worked with github action.
But it looks like you created the cron schedule, so please solve it (or remove it) before everybody gets annoyed.

nashif, 2020-07-01:
Go to Settings > Actions > Actions permissions to Disable Actions for the repository in question your fork.

closed 2020-07-01 by nashif

ABOSTM, 2020-07-01:
@nashif thanks for the hint.