zephyrproject-rtos/zephyr #24226: [master]Bluetooth: samples/bluetooth/central_hr can't connect with samples/bluetooth/peripheral_hr

Jeepgoing, 2020-04-09:
Platform:
- Central: nrf52dk_nrf52832
- Peripheral: thingy52_nrf52832

Environment:
- OS: (Ubuntu 18.04)
- Toolchain (Zephyr SDK)

Issue:
I am testing BLE, thingy52_nrf52832 as peripheral and nrf52dk_nrf52832 ad central, it's sames good about peripheral, when i download `samples/bluetooth/peripheral_hr` to thingy52_nrf52832, i can connect it with my Android App[nRF Connect], but when i use thingy52_nrf52832 as central `samples/bluetooth/central_hr`, the log show central can't connect with peripheral, the central always stay at https://github.com/zephyrproject-rtos/zephyr/blob/6540f81f7ca2859d50a9c7d9b216ee5da04bd291/samples/bluetooth/central_hr/src/main.c#L67-L68 can't go to https://github.com/zephyrproject-rtos/zephyr/blob/6540f81f7ca2859d50a9c7d9b216ee5da04bd291/samples/bluetooth/central_hr/src/main.c#L79-L82

carlescufi, 2020-04-09:
This works fine here for me, I am not sure what the issue is

carlescufi, 2020-04-09:
Here is my log from the central side:

```
*** Booting Zephyr OS build zephyr-v2.2.0-1321-g3a25c4544bb5  ***
Bluetooth initialized
Scanning successfully started
[00:00:00.006,530] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.006,530] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.006,561] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 2.2 Build 99
[00:00:00.007,171] <inf> bt_hci_core: Identity: e0:79:10:2c:fc:f3 (random)
VICE]: 52:45:92:79:e8:d2 (random), AD evt type 0, AD data len 17, RSSI -61
[AD]: 1 data_len 1
[AD]: 10 data_len 1
[AD]: 255 data_len 9
[DEVICE]: 52:45:92:79:e8:d2 (random), AD evt type 4, AD data len 0, RSSI -62
[DEVICE]: 4c:6a:62:e8:17:8f (random), AD evt type 0, AD data len 18, RSSI -35
[AD]: 1 data_len 1
[AD]: 10 data_len 1
[AD]: 255 data_len 10
[00:00:00.007,171] <inf> bt_hci_core: HCI: version 5.1 (0x0a) revision 0x0000, manufacturer 0x05f1
[DEVICE]: 09:58:fb:c5:7b:72 (random), AD evt type 3, AD data len 15, RSSI -54
[00:00:00.007,171[DEVICE]: c8:7a:f3:8d:12:7b (random), AD evt type 0, AD data len 11, RSSI -30
[AD]: 1 data_len 1
[AD]: 3 data_len 6
] <inf> bt_hci_core: LMP: version 5.1 (0x0a) subver 0xffff
Connected: c8:7a:f3:8d:12:7b (random)
[ATTRIBUTE] handle 25
[ATTRIBUTE] handle 26
[ATTRIBUTE] handle 28
[SUBSCRIBED]
[NOTIFICATION] data 0x20004c33 length 2
[NOTIFICATION] data 0x20004c33 length 2
[NOTIFICATION] data 0x20004c33 length 2
```
Maybe your central is connecting to another device that also shows the HR service in its advertising data, and that device is not handling the ATT requests correctly?

Jeepgoing, 2020-04-10:
I have updated master branch and tried again, it's the same result：

central log：

> *** Booting Zephyr OS build zephyr-v2.2.0-1364-gab15602b7964  ***
> Bluetooth initialized
> Scanning successfully started
> [00:00:00.006,622] [0m<inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)[0m
> [00:00:00.006,622] [0m<inf> bt_hci_core: HW Variant: nRF52x (0x0002)[0m
> VICE]: 17:50:47:3e:65:18 (random), AD evt type 3, AD data len 31, RSSI -71
> [DEVICE]: de:12:1c:d9:9a:e9 (random), AD evt type 0, AD data len 11, RSSI -38
> [AD]: 1 data_len 1
> [AD]: 3 data_len 6
> [00:00:00.006,622] [0m<inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 2.2 Build 99[0m
> [00:00:00.007,293] [0m<inf> bt_hci_core: Identity: de:14:0c:a4:07:b6 (random)[0m
> [00:00:00.007,324] [0m<inf> bt_hci_core: HCI: version 5.1 (0x0a) revision 0x0000, manufacturer 0x05f1[0m
> [00:00:00.007,324] [0m<inf> bt_hci_core: LMP: version 5.1 (0x0a) subver 0xffff[0m
> Connected: de:12:1c:d9:9a:e9 (random)
> [ATTRIBUTE] handle 25
> uuid = 180d
> [ATTRIBUTE] handle 26
> uuid = 2a37
> Discover complete
> Disconnected: de:12:1c:d9:9a:e9 (random) (reason 0x08)
> Scanning successfully started
> [DEVICE]: 38:08:56:b5:69:27 (random), AD evt type 3, AD data len 31, RSSI -79
> [DEVICE]: e6:30:38:f2:8d:f1 (random), AD evt type 0, AD data len 28, RSSI -72
> [DEVICE]: 1e:63:f2:e3:a6:d1 (random), AD evt type 3, AD data len 31, RSSI -71
> [DEVICE]: 26:94:19:bc:10:da (random), AD evt type 3, AD data len 31, RSSI -71
> [DEVICE]: 17:50:47:3e:65:18 (random), AD evt type 3, AD data len 31, RSSI -66
> [DEVICE]: 4f:18:df:b6:38:20 (random), AD evt type 0, AD data len 17, RSSI -78
> [DEVICE]: 4f:18:df:b6:38:20 (random), AD evt type 4, AD data len 0, RSSI -77
> [DEVICE]: 2a:32:f5:10:96:de (random), AD evt type 3, AD data len 31, RSSI -76
> [DEVICE]: 15:85:bb:f8:e7:37 (random), AD evt type 3, AD data len 31, RSSI -56
> [DEVICE]: 65:5d:ea:7f:7d:fd (random), AD evt type 0, AD data len 17, RSSI -77
> [DEVICE]: 13:fc:ba:23:cb:83 (random), AD evt type 3, AD data len 31, RSSI -77
> [DEVICE]: 09:84:ef:11:93:4c (random), AD evt type 3, AD data len 31, RSSI -68
> [DEVICE]: 15:85:bb:f8:e7:37 (random), AD evt type 3, AD data len 31, RSSI -70
> [DEVICE]: 17:01:cc:4b:c1:6f (random), AD evt type 3, AD data len 31, RSSI -77
> [DEVICE]: e6:30:38:f2:8d:f1 (random), AD evt type 0, AD data len 28, RSSI -72
> [DEVICE]: 5a:f8:4a:2a:31:21 (random), AD evt type 0, AD data len 18, RSSI -73
> [DEVICE]: 09:31:6a:ce:b2:3c (random), AD evt type 3, AD data len 31, RSSI -70
> [DEVICE]: 49:ad:75:03:eb:ba (random), AD evt type 0, AD data len 18, RSSI -69
> [DEVICE]: 75:2b:70:46:b5:11 (random), AD evt type 0, AD data len 18, RSSI -60
> [DEVICE]: 75:2b:70:46:b5:11 (random), AD evt type 4, AD data len 0, RSSI -60
> [DEVICE]: 1e:83:a8:1d:a0:eb (random), AD evt type 3, AD data len 31, RSSI -72
> [DEVICE]: 2f:63:8a:5f:59:bf (random), AD evt type 3, AD data len 31, RSSI -73
> [DEVICE]: de:12:1c:d9:9a:e9 (random), AD evt type 0, AD data len 11, RSSI -36
> [AD]: 1 data_len 1
> [AD]: 3 data_len 6
> Connected: de:12:1c:d9:9a:e9 (random)
> [ATTRIBUTE] handle 25
> [ATTRIBUTE] handle 26

peripheral log:

> Disconnected (reason 0x08)
> Connected
> Disconnected (reason 0x08)
> Connected

I have tested peripheral with my phone use nRF Toolbox APP, i have't found another HRM device, only my device was found, so i don't know why your result is good. @carlescufi

Jeepgoing, 2020-04-11:
I take the board to my home and change the central side from nrf52dk_nrf52832 to thingy52_nrf52832, 
means i test with two thingy52_nrf52832 borad, it's sames ok.

logs:
```
[10:42:00.697]收←◆*** Booting Zephyr OS build zephyr-v2.2.0-1378-g071a986bb676  ***
Bluetooth initialized
Scanning successfully started

[10:42:00.791]收←◆[DEVICE]: de:12:1c:d9:9a:e9 (random), AD evt type 0, AD data len 11, RSSI -49
[AD]: 1 data_len 1
[AD]: 3 data_len 6

[10:42:01.103]收←◆Connected: de:12:1c:d9:9a:e9 (random)

[10:42:01.197]收←◆[ATTRIBUTE] handle 25

[10:42:01.400]收←◆[ATTRIBUTE] handle 26

[10:42:01.681]收←◆[00:00:00.004,882] [0m<inf> sx1509b: GPIO_P0 init ok[0m
[00:00:00.012,207] [0m<inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)[0m
[00:00:00.012,237] [0m<inf> bt_hci_core: HW Variant: nRF52x (0x0002)[0m
[00:00:00.012,237] [0m<inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 2.2 Build 99[0m
[00:00:00.012,847] [0m<inf> bt_hci_core: Identity: c4:57:62:f0:f2:8a (random)[0m
[00:00:00.012,847] [0m<inf> bt_hci_core: HCI: version 5.1 (0x0a) revision 0x0000, manufacturer 0x05f1[0m
[00:00:00.012,847] [0m<inf> bt_hci_core: LMP: version 5.1 (0x0a) subver 0xffff[0m
[ATTRIBUTE] handle 28
[SUBSCRIBED]

[10:42:02.650]收←◆[NOTIFICATION] data 0x200047b3 length 2

[10:42:03.647]收←◆[NOTIFICATION] data 0x200047b3 length 2

[10:42:04.647]收←◆[NOTIFICATION] data 0x200047b3 length 2
```

Jeepgoing, 2020-04-11:
@carlescufi Hi Can you show how to send message from central to peripheral? i can't find samples on master branch, thanks.

Jeepgoing, 2020-04-13:
@carlescufi which board are you using? I have tested nrf52dk_nrf52832 and thingy52_nrf52832 with the same sample `samples/bluetooth/peripheral_hr` but the result is not same, i can connect with thingy52_nrf52832 very smoothly, but can't connect nrf52dk_nrf52832.

nrf52dk_nrf52832 log:
```
*** Booting Zephyr OS build zephyr-v2.2.0-1416-g47f1665e7837  ***
Bluetooth initialized
[00:00:00.006,652] [0m<inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)[0m
[00:00:00.006,652] [0m<inf> bt_hci_core: HW Variant: nRF52x (0x0002)[0m
[00:00:00.006,652] [0m<inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 2.2 Build 99[0m
Advertising successfully started
[00:00:00.007,324] [0m<inf> bt_hci_core: Identity: de:14:0c:a4:07:b6 (random)[0m
[00:00:00.007,354] [0m<inf> bt_hci_core: HCI: version 5.1 (0x0a) revision 0x0000, manufacturer 0x05f1[0m
[00:00:00.007,354] [0m<inf> bt_hci_core: LMP: version 5.1 (0x0a) subver 0xffff[0m
Connected
Disconnected (reason 0x08)
```

thingy52_nrf52832 log:
```
*** Booting Zephyr OS build zephyr-v2.2.0-1416-g47f1665e7837  ***
Bluetooth initialized
Advertising successfully started
[00:00:00.004,791] [0m<inf> sx1509b: GPIO_P0 init ok[0m
[00:00:00.012,115] [0m<inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)[0m
[00:00:00.012,115] [0m<inf> bt_hci_core: HW Variant: nRF52x (0x0002)[0m
[00:00:00.012,145] [0m<inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 2.2 Build 99[0m
[00:00:00.012,756] [0m<inf> bt_hci_core: Identity: c4:57:62:f0:f2:8a (random)[0m
[00:00:00.012,756] [0m<inf> bt_hci_core: HCI: version 5.1 (0x0a) revision 0x0000, manufacturer 0x05f1[0m
[00:00:00.012,756] [0m<inf> bt_hci_core: LMP: version 5.1 (0x0a) subver 0xffff[0m
Connected
[00:00:07.826,263] [0m<inf> hrs: HRS notifications enabled[0m
```

Jeepgoing, 2020-04-14:
> @carlescufi Hi Can you show how to send message from central to peripheral? i can't find samples on master branch, thanks.

I know how to send message from central to peripheral

carlescufi, 2020-04-14:
Closing since this seems solved

closed 2020-04-14 by carlescufi