zephyrproject-rtos/zephyr #24227: build hello_world sample failed for ESP32 board.

awsa2ron, 2020-04-09:
**Describe the bug**
I can build successfully with this [ESP32 official instruction](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#get-started-setup-toochain)
But failed with zephyr.

**To Reproduce**
`west build -b esp32 samples/hello_world`

**Screenshots or console output**
```
$ west build -b esp32 samples/hello_world
[9/35] Building C object zephyr/drivers/i2c/CMakeFiles/drivers__i2c.dir/i2c_esp32.c.obj
FAILED: zephyr/drivers/i2c/CMakeFiles/drivers__i2c.dir/i2c_esp32.c.obj 
ccache /home/aaron/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DBUI
LD_VERSION=zephyr-v2.2.0-1276-g6540f81f7ca2 -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I../
include -Izephyr/include/generated -I../soc/xtensa/esp32 -I../soc/xtensa/esp32/include -I/home/aaron/repos/zephyrproj
ect/modules/hal/esp-idf/zephyr/../components/esp32/include -I/home/aaron/repos/zephyrproject/modules/hal/esp-idf/zeph
yr/../components/soc/esp32/include -isystem ../lib/libc/minimal/include -isystem /home/aaron/.espressif/tools/xtensa-
esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/include -isystem /home/aaron/.espre
ssif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/include-fixed -Os
 -imacros/home/aaron/repos/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common 
-g -imacros/home/aaron/repos/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security 
-Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int
 -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro
-prefix-map=/home/aaron/repos/zephyrproject/zephyr/samples/hello_world=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/aaro
n/repos/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/aaron/repos/zephyrproject=WEST_TOPDIR -ffunction-se
ctions -fdata-sections -mlongcalls -std=c99 -nostdinc -MD -MT zephyr/drivers/i2c/CMakeFiles/drivers__i2c.dir/i2c_esp3
2.c.obj -MF zephyr/drivers/i2c/CMakeFiles/drivers__i2c.dir/i2c_esp32.c.obj.d -o zephyr/drivers/i2c/CMakeFiles/drivers
__i2c.dir/i2c_esp32.c.obj   -c /home/aaron/repos/zephyrproject/zephyr/drivers/i2c/i2c_esp32.c
/home/aaron/repos/zephyrproject/zephyr/drivers/i2c/i2c_esp32.c: In function 'i2c_esp32_transmit':
/home/aaron/repos/zephyrproject/zephyr/drivers/i2c/i2c_esp32.c:34:30: error: incompatible type for argument 2 of 'k_s
em_take'
 #define I2C_ESP32_TIMEOUT_MS 100
                              ^~~
/home/aaron/repos/zephyrproject/zephyr/drivers/i2c/i2c_esp32.c:289:34: note: in expansion of macro 'I2C_ESP32_TIMEOUT
_MS'
  if (k_sem_take(&data->fifo_sem, I2C_ESP32_TIMEOUT_MS) < 0) {
                                  ^~~~~~~~~~~~~~~~~~~~
In file included from ../include/kernel.h:5700,
                 from ../include/device.h:10,
                 from ../include/drivers/gpio.h:23,
                 from /home/aaron/repos/zephyrproject/zephyr/drivers/i2c/i2c_esp32.c:17:
zephyr/include/generated/syscalls/kernel.h:746:62: note: expected 'k_timeout_t' {aka 'struct <anonymous>'} but argument is of type 'int'
 static inline int k_sem_take(struct k_sem * sem, k_timeout_t timeout)
                                                  ~~~~~~~~~~~~^~~~~~~
[14/35] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mutex.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/aaron/repos/zephyrproject/zephyr/build
```

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain (Zephyr SDK, ...)
 - Commit SHA：6540f81f7ca2859d50a9c7d9b216ee5da04bd291

tejlmand, 2020-04-09:
@awsa2ron it appears to my that you are not using Zephyr SDK, but a 3rd party toolchain from Espressif for compilation, is that correct ?
If so, please update the text, describing the toolchain you are using.

In above log, it shows:
> .espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc

awsa2ron, 2020-04-09:
@tejlmand 
Thanks for your reply!
I just follow the [zephyr official docs](https://docs.zephyrproject.org/latest/boards/xtensa/esp32/doc/index.html#set-up-build-environment)

I think zephyr need using 3rd party toolchain in this case.

tejlmand, 2020-04-09:
@carlescufi as the official Zephyr docs that @awsa2ron is pointing to, tells users to try hello world with the Espressif toolchain, then I think this is indeed a bug. Re-adding the label 'bug'.

tejlmand, 2020-04-09:
Please note, this is not Zephyr SDK, but 3rd party toolchain as described in the Zephyr doc link.

swapnabulbule, 2020-04-11:
@awsa2ron I was getting same error.  So, I checked any issue is open on I2C for esp32. [issue #24266](https://github.com/zephyrproject-rtos/zephyr/pull/24266)
As per [this](https://github.com/zephyrproject-rtos/zephyr/commit/e9c5e4824c8c600a4b7f7d2c7643a4b241761405) commit . I modified the code. And it worked.

jenmwms, 2020-04-21:
> @carlescufi as the official Zephyr docs that @awsa2ron is pointing to, tells users to try hello world with the Espressif toolchain, then I think this is indeed a bug. 

Perhaps it would help to state in the Zephyr doc to follow the section on "Setting up Development Environment", then return to the Zephyr doc? https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#setting-up-development-environment

nashif, 2020-04-21:
fixed by commit e9c5e4824c8c600a4b7f7d2c7643a4b241761405

closed 2020-04-21 by nashif