zephyrproject-rtos/zephyr #24233: adxl362_trigger.c adxl362_init_interrupt function :const struct adxl362_config *cfg  not found gpio_cs_port

ytchen1435, 2020-04-09:
because adxl362_trigger.c not define DT_DRV_COMPAT

pabigot, 2020-04-14:
What platform are you building this on?  I can't see any dependency between the trigger module and the chip-select GPIOS, and I can't duplicate a build failure.

ytchen1435, 2020-04-20:
#24421 pull request fix

pabigot, 2020-04-20:
That added the define in the main routine, where it's needed.  I need you to tell me what target platform you're building for that does not build if it isn't also added in the trigger source, which AFAICT does not need it.

`samples/sensor/adxl362` which enables triggers builds correctly for me for both nrf52dk_nrf52832 and frdm_k64f.

pabigot, 2020-06-29:
Closing as the suggested fix, though apparently unrelated, has been merged and no problem can be reproduced.  Open a new issue if this recurs.

closed 2020-06-29 by pabigot