zephyrproject-rtos/zephyr #24237: Fail to pass samples/subsys/nvs

ABOSTM, 2020-04-09:
**Describe the bug**
I tried to run samples/subsys/nvs on disco_l475_iot,
but after some reboots (expected  in this test),
it fails to read :

> Id: 4, Data: DATA

**To Reproduce**
Steps to reproduce the behavior:
1. west -v build -p auto -b disco_l475_iot1 samples/subsys/nvs/ 
   west flash
2. Open console
3. after 10 cycles, ID 4 is missing in the log

**console:**
> Rebooting in ...5...4...3...2...1
> *** Booting Zephyr OS build zephyr-v2.2.0-1251-gb06a5b57078a  ***
> Id: 1, Address: 192.168.1.1
> Id: 2, Key: ff fe fd fc fb fa f9 f8
> Id: 3, Reboot_counter: 10
> **Id: 4, Data: DATA**
> Id: 5, Longarray: 0 1 2 3 4 5 6 7 8 9 a b c d e f 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f
> [00:00:00.009,000] <inf> fs_nvs: 3 Sectors of 2048 bytes
> [00:00:00.009,000] <inf> fs_nvs: alloc wra: 0, 778
> Reboot counter history: ...10...9...8...7...6...5...4...3...2...1...0
> Oldest reboot counter: 0
> Rebooting in ...5...4...3...2...1
> *** Booting Zephyr OS build zephyr-v2.2.0-1251-gb06a5b57078a  ***
> Id: 1, Address: 192.168.1.1
> Id: 2, Key: ff fe fd fc fb fa f9 f8
> Id: 3, Reboot_counter: 11
> Id: 5, Longarray: 0 1 2 3 4 5 6 7 8 9 a b c d e f 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f 

**Preanalysis:**
Analysing this issue, I found that function `nvs_read_hist() `returns with` -ENOENT`. But then I am a bit lost with aet management.
The same tests fails on **nucleo_l4r5zi** and **stm32f3_disco**.
Also @erwango tested board: **frdm_k64f** and test is failed too, but log is slightly different (see below).

> *** Booting Zephyr OS build zephyr-v2.2.0-1124-g4cc25ce22efa  ***
> Flash Init failed
> No address found, adding 192.168.1.1 at id 1
> No key found, adding it at id 2
> No Reboot counter found, adding it at id 3
> Id: 4 not found, adding it
> Longarray not found, adding it as id 5
> [00:00:00.007,000] <err> fs_nvs: NVS not initialized
> [00:00:00.011,000] <err> fs_nvs: NVS not initialized
> [00:00:00.011,000] <err> fs_nvs: NVS not initialized
> [00:00:00.014,000] <err> fs_nvs: NVS not initialized
> [00:00:00.014,000] <err> fs_nvs: NVS not initialized
> [00:00:00.018,000] <err> fs_nvs: NVS not initialized
> [00:00:00.018,000] <err> fs_nvs: NVS not initialized
> [00:00:00.020,000] <err> fs_nvs: NVS not initialized
> [00:00:00.020,000] <err> fs_nvs: NVS not initialized
> [00:00:00.023,000] <err> fs_nvs: NVS not initialized
> Reboot counter history:
> Error, no Reboot counter

**Environment :**
 - OS: Windows
 - Toolchain: Zephyr SDK
 - Commit SHA : b06a5b57078a4e8e09c4e6cb983ae6c1cef08ea2

carlescufi, 2020-04-09:
@Laczen any hints on this one?

Laczen, 2020-04-09:
@ABOSTM, the bug is not a bug, it is supposed to do this. After 10 cycles the item with ID=4 is deleted, so it should show anymore.

Regarding the error on the frdm_k64f, I think this device has a pagesize of 0x1000, so only two pages will fit in the storage area. The test requires 3 pages so the initialisation will fail.

erwango, 2020-04-10:
@Laczen if not failing, is there any way to get a "passed" verdict when running the sample with sanitycheck?

Laczen, 2020-04-10:
@erwango, this is a sample to help people gettings started with nvs.  This sample is no test whatsoever so there should never be any conclusion about a run with sanitycheck.

Laczen, 2020-04-10:
@erwango, if there is a need to get this "passed" verdict, please feel free to remove the sample.

erwango, 2020-04-10:
@Laczen this is not helping

Laczen, 2020-04-11:
@erwango, OK, let us restart. The nvs sample is a sample, as many samples it is not really usefull to run it by sanitycheck. But suppose it is run by sanitycheck, what would you like to conclude from this? I hope you would not conclude that everything is OK if you are getting the same result as a preprogrammed result. The output of the sample is dependent on the configuration of the storage area (the sector size, the page size, the write block size, ...) and these are soc dependent. The sample could easily be changed to stop after 10 reboots and to check that the delete is properly performed. But at that moment there has been no 'garbage collection', after how many moments the 'garbage collection' is performed is configuration dependent. Stopping after 10 reboots is clearly not an option.

akofoed, 2020-04-14:
Hi, @Laczen and @erwango 
I have tried the NVS sample on a nucleo-F411RE and nucleo-429zi and I get the same as @ABOSTM gets; "No xxx found... Adding..." Rebooting...

I have tried to do as @erwango in #19895 setting flash region to 0x100000 (1M) and size to 0x8000 but no matter what I try it either gives me an invalid flash configuration (which I then disabled in nvs.c to see if it would help, but no) or that nvs_write() fails with error: -5

Let me know if you want me to test something.

Laczen, 2020-04-14:
Hi @akofoed, on the nucleo-F411RE and nucleo-429zi is the first line "Flash Init failed" ?

For the nvs sample to work the following conditions must be OK:
a. The storage area should be defined correctly on page boundaries,
b. The nvs sector size should be a multiple of the page size (can be 1),
c. The nvs sector size (and thus also the page size) should be smaller than 32kB,
d. There should at least 3 sectors in the storage area.

Typical examples that will not work:
a. Trying to define the storage area in one 64kB page,
b. Trying to define the storage area in one (or multiple) 128kB page(s).

Do the nucleo-F411RE and nucleo-429zi meet all required conditions ?

akofoed, 2020-04-14:
for the nucleo_429zi, I get the following output (zephyr master updated a few days ago):
Using the 64KB sector @10000 (as in the current nucleo_f429zi.dts) understandingly does not work as NVS requires at least two sectors to work. But using 2 (or 3 or 4) 16KB sectors at 0x100000 should work but it does not:
```
Rebooting in ...5...4...3...2...1�*** Booting Zephyr OS build zephyr-v2.2.0-1102-gf57ec57ce557  ***
No address found, adding 192.168.1.1 at id 1
No key found, adding it at id 2
No Reboot counter found, adding it at id 3
Id: 4 not found, adding it
Longarray not found, adding it as id 5
[00:00:00.016,000] <inf> fs_nvs: 3 Sectors of 16384 bytes
[00:00:00.016,000] <inf> fs_nvs: alloc wra: 0, 3ff0
[00:00:00.016,000] <inf> fs_nvs: data wra: 0, 0
Reboot counter history: 
 Error, no Reboot counter
Rebooting in ...5...4...3...2...1�*** Booting Zephyr OS build zephyr-v2.2.0-1102-gf57ec57ce557  ***
No address found, adding 192.168.1.1 at id 1
...
...
```
The following DTS storage_partition is used:
`
		storage_partition: partition@100000 {
			label = "storage";
			reg = <0x000100000 0x0008000>;
		};
`

Laczen, 2020-04-14:
Hi @akofoed, this is strange, it was reported to work correctly in #19895.

Your setup is strange, it is reporting that there are 3 sectors while your dts definition only has 2. Could you please undo any changes that are ignoring errors and try again.

akofoed, 2020-04-14:
The DTS only says where and how much, not sector count.
I am running the stock sample that has hard-coded 3 sectors into it:
https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/subsys/nvs/src/main.c#L83
However, the STM32F429ZI does have 4 sectors with 16KB starting at 0x100000 (1M) so all should be fine.

Laczen, 2020-04-14:
> The DTS only says where and how much, not sector count.
> I am running the stock sample that has hard-coded 3 sectors into it: [...]

This is not exactly true, the dts says size is 32kB, the page info says that these are 16kB sectors, this means that you can only have two sectors in there.
If you would say that the size is C000 you would have sufficient size for 3 sectors.

akofoed, 2020-04-14:
Ups, you are correct. I have now changed the DTS size to 64KB (all 4 sectors):
```
storage_partition: partition@100000 {
			label = "storage";
			reg = <0x000100000 0x00010000>;
		};
```

Output is still the same:
```
Rebooting in ...5...4...3...2...1�*** Booting Zephyr OS build zephyr-v2.2.0-1102-gf57ec57ce557  ***
No address found, adding 192.168.1.1 at id 1
No key found, adding it at id 2
No Reboot counter found, adding it at id 3
Id: 4 not found, adding it
Longarray not found, adding it as id 5

[00:00:00.016,000] <inf> fs_nvs: 3 Sectors of 16384 bytes
[00:00:00.016,000] <inf> fs_nvs: alloc wra: 0, 3ff0
[00:00:00.016,000] <inf> fs_nvs: data wra: 0, 0
Reboot counter history: 
 Error, no Reboot counter
Rebooting in ...5...4...3...2...1�*** Booting Zephyr OS build zephyr-v2.2.0-1102-gf57ec57ce557  ***
No address found, adding 192.168.1.1 at id 1
...
...
```

Laczen, 2020-04-14:
> Ups, you are correct. I have now changed the DTS size to 64KB (all 4 sectors):
>
> ```
> storage_partition: partition@100000 {
> 			label = "storage";
> 			reg = <0x000100000 0x00010000>;
> 		};
> ```
>
> Output is [...]

Please undo changes that are hiding init errors. I can only guess that there is something wrong because it is hidden. Probably (but this is a guess) there was already some data in the flash region and it needs to be cleared before first use, but then again this is only a guess.

akofoed, 2020-04-14:
Sure. I have now done a full-chip-erase (STM32CubeProgrammer) and this is the output below just after doing a "west flash" using minicom as term.
It seems that the LOG_INF from nvs_init() is printed after the printfk() statements. 
If there is something you want me to try, just let me know. I have some other STM32 boards like STM32L476 and STM32L401/411 lying around. Thanks

```
Port /dev/ttyACM0, 12:43:12

Press CTRL-A Z for help on special keys

*** Booting Zephyr OS build zephyr-v2.2.0-1102-gf57ec57ce557  ***
No address found, adding 192.168.1.1 at id 1
No key found, adding it at id 2
No Reboot counter found, adding it at id 3
Id: 4 not found, adding it
Longarray not found, adding it as id 5
[00:00:00.016,000] <inf> fs_nvs: 3 Sectors of 16384 bytes
[00:00:00.016,000] <inf> fs_nvs: alloc wra: 0, 3ff0
[00:00:00.016,000] <inf> fs_nvs: data wra: 0, 0
Reboot counter history: 
 Error, no Reboot counter
Rebooting in ...5...4...3...2...1�*** Booting Zephyr OS build zephyr-v2.2.0-1102-gf57ec57ce557  ***
No address found, adding 192.168.1.1 at id 1
No key found, adding it at id 2
No Reboot counter found, adding it at id 3
Id: 4 not found, adding it
Longarray not found, adding it as id 5
[00:00:00.016,000] <inf> fs_nvs: 3 Sectors of 16384 bytes
[00:00:00.016,000] <inf> fs_nvs: alloc wra: 0, 3ff0
```

Laczen, 2020-04-14:
It seems nothing is written at all (alloc wra does not change). I think the board definition for the STM32F429ZI is missing some vital information. As I do not have the these boards it is impossible for me to find out exactly what is wrong. I think it is related to #23997.

Laczen, 2020-04-14:
@akofoed, I think it is because the stm32f429zi is missing a `compatible = "st,stm32l4-flash-controller"` line in the dtsi definition.

akofoed, 2020-04-14:
The DTS for the 429zi do inherit the compatible="st,stm32f4-flash-controller" from the top SOC DTS file (stm32f4.dtsi).

In the mean time I found out that for the STM32F4, the write-block-size= < 1 > and erase-block-size is missing. This is not the case for the STM32L476 (which correctly inherits  write-block-size = < 8 > and erase-block-size = < 2048 >), but running it on the nucleo_l476rg board using this overlay file:
```
&flash0 {
	/*
	 * For more information, see:
	 * http://docs.zephyrproject.org/devices/dts/flash_partitions.html
	 */
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		/* application image slot: 256KB */
		slot0_partition: partition@00000 {
			label = "image-0";
			reg = <0x00000000 0x00020000>;
		};

		/* backup slot: 256KB */
		slot1_partition: partition@20000 {
			label = "image-1";
			reg = <0x00020000 0x00020000>;
		};

		/* swap slot: 128KB */
		scratch_partition: partition@40000 {
			label = "image-scratch";
			reg = <0x00040000 0x00020000>;
		};

		/* storage: 64KB for settings */
		storage_partition: partition@60000 {
			label = "storage";
			reg = <0x00060000 0x00020000>;
		};

	};
};
```

Still does not work (run after mass erase):

```
Port /dev/ttyACM0, 17:34:28

Press CTRL-A Z for help on special keys

*** Booting Zephyr OS build zephyr-v2.2.0-1102-gf57ec57ce557  ***
No address found, adding 192.168.1.1 at id 1
No key found, adding it at id 2
No Reboot counter found, adding it at id 3
Id: 4 not found, adding it
Longarray not found, adding it as id 5
[00:00:00.007,000] <inf> fs_nvs: 3 Sectors of 2048 bytes
[00:00:00.007,000] <inf> fs_nvs: alloc wra: 0, 7f0
[00:00:00.007,000] <inf> fs_nvs: data wra: 0, 0
Reboot counter history: 
 Error, no Reboot counter
*** Booting Zephyr OS build zephyr-v2.2.0-1102-gf57ec57ce557  ***
No address found, adding 192.168.1.1 at id 1
No key found, adding it at id 2
No Reboot counter found, adding it at id 3
Id: 4 not found, adding it
Longarray not found, adding it as id 5
```

Laczen, 2020-04-14:
@akofoed, for the 429zi there is no fixed erase block size, that is why you can't find it.

Could you check that the error (of each write) is still -5 (I was only guessing).
Could you also check if this error was also present in zephyr 2.2.

akofoed, 2020-04-14:
The nucleo_l476 says -5 modifying main.c line 102 to:
```
                int rc = nvs_write(&fs, ADDRESS_ID, &buf, strlen(buf)+1);
                printk("nvs_write rc: %d\n", rc);
```

Gives -5 adding 192.168.1.1:

```
Port /dev/ttyACM0, 18:17:13

Press CTRL-A Z for help on special keys

*** Booting Zephyr OS build zephyr-v2.2.0-1102-gf57ec57ce557  ***
No address found, adding 192.168.1.1 at id 1
nvs_write rc: -5
No key found, adding it at id 2
No Reboot counter found, adding it at id 3
Id: 4 not found, adding it
Longarray not found, adding it as id 5
[00:00:00.007,000] <inf> fs_nvs: 3 Sectors of 2048 bytes
[00:00:00.007,000] <inf> fs_nvs: alloc wra: 0, 7f0
[00:00:00.007,000] <inf> fs_nvs: data wra: 0, 0
Reboot counter history: 
 Error, no Reboot counter
*** Booting Zephyr OS build zephyr-v2.2.0-1102-gf57ec57ce557  ***
No address found, adding 192.168.1.1 at id 1
nvs_write rc: -5
No key found, adding it at id 2
No Reboot counter found, adding it at id 3
Id: 4 not found, adding it
Longarray not found, adding it as id 5
[00:00:00.007,000] <inf> fs_nvs: 3 Sectors of 2048 bytes
[00:00:00.007,000] <inf> fs_nvs: alloc wra: 0, 7f0
[00:00:00.007,000] <inf> fs_nvs: data wra: 0, 0
```

Laczen, 2020-04-14:
@akofoed, could be that there is something wrong with the pagesizes that are returned from the flash driver. In the sample the sectorsize is retrieved from the pagesize (line 82) could you print the sectorsize ?

Laczen, 2020-04-14:
@akofoed, the error -5 (-EIO) is coming directly from the flash driver.

akofoed, 2020-04-14:
Interesting... I checked out zephyr-v2.2.0 as you suggested and it works on the nucleo_l476:

```
*** Booting Zephyr OS build zephyr-v2.2.0  ***
No address found, adding 192.168.1.1 at id 1
No key found, adding it at id 2
No Reboot counter found, adding it at id 3
Id: 4 not found, adding it
Longarray not found, adding it as id 5
[00:00:00.005,000] <inf> fs_nvs: 3 Sectors of 2048 bytes
[00:00:00.005,000] <inf> fs_nvs: alloc wra: 0, 7f0
[00:00:00.005,000] <inf> fs_nvs: data wra: 0, 0
Reboot counter history: ...0
Oldest reboot counter: 0
Rebooting in ...5...4...3...2...1
*** Booting Zephyr OS build zephyr-v2.2.0  ***
Id: 1, Address: 192.168.1.1
Id: 2, Key: ff fe fd fc fb fa f9 f8 
Id: 3, Reboot_counter: 1
Id: 4, Data: DATA
Id: 5, Longarray: 0 1 2 3 4 5 6 7 8 9 a b c d e f 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 2 
[00:00:00.005,000] <inf> fs_nvs: 3 Sectors of 2048 bytes
[00:00:00.005,000] <inf> fs_nvs: alloc wra: 0, 7c0
[00:00:00.005,000] <inf> fs_nvs: data wra: 0, b0
Reboot counter history: ...1...0
Oldest reboot counter: 0
Rebooting in ...5...4...3...2...1
*** Booting Zephyr OS build zephyr-v2.2.0  ***
Id: 1, Address: 192.168.1.1
Id: 2, Key: ff fe fd fc fb fa f9 f8 
Id: 3, Reboot_counter: 2
Id: 4, Data: DATA
```

Laczen, 2020-04-14:
OK, you could try the same with the stm32f429zi (with the correct dts or overlay file).

akofoed, 2020-04-14:
zephyr-v2.2.0 works on the F429 (nucleo_f429zi):
```
Port /dev/ttyACM0, 18:45:22

Press CTRL-A Z for help on special keys

*** Booting Zephyr OS build zephyr-v2.2.0  ***
No address found, adding 192.168.1.1 at id 1
No key found, adding it at id 2
No Reboot counter found, adding it at id 3
Id: 4 not found, adding it
Longarray not found, adding it as id 5
[00:00:00.014,000] <inf> fs_nvs: 3 Sectors of 16384 bytes
[00:00:00.014,000] <inf> fs_nvs: alloc wra: 0, 3ff0
[00:00:00.014,000] <inf> fs_nvs: data wra: 0, 0
Reboot counter history: ...0
Oldest reboot counter: 0
Rebooting in ...5...4...3...2...1
*** Booting Zephyr OS build zephyr-v2.2.0  ***
Id: 1, Address: 192.168.1.1
Id: 2, Key: ff fe fd fc fb fa f9 f8 
Id: 3, Reboot_counter: 1
Id: 4, Data: DATA
Id: 5, Longarray: 0 1 2 3 4 5 6 7 8 9 a b c d e f 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 2 
[00:00:00.016,000] <inf> fs_nvs: 3 Sectors of 16384 bytes
[00:00:00.016,000] <inf> fs_nvs: alloc wra: 0, 3fc0
[00:00:00.016,000] <inf> fs_nvs: data wra: 0, a1
Reboot counter history: ...1...0
Oldest reboot counter: 0
Rebooting in ...5...4...3...2...1
*** Booting Zephyr OS build zephyr-v2.2.0  ***
Id: 1, Address: 192.168.1.1
Id: 2, Key: ff fe fd fc fb fa f9 f8 
```
However, I have modified the flash area:
```
		/* storage: 64KB for settings */
		storage_partition: partition@100000 {
			label = "storage";
			reg = <0x00100000 0x00010000>;
		};
```
Else with default DTS I get:
```
Press CTRL-A Z for help on special keys

*** Booting Zephyr OS build zephyr-v2.2.0  ***
Flash Init failed
No address found, adding 192.168.1.1 at id 1
No key found, adding it at id 2
No Reboot counter found, adding it at id 3
Id: 4 not found, adding it
Longarray not found, adding it as id 5
[00:00:00.004,000] <err> fs_nvs: Invalid sector size
[00:00:00.005,000] <err> fs_nvs: NVS not initialized
[00:00:00.009,000] <err> fs_nvs: NVS not initialized
[00:00:00.009,000] <err> fs_nvs: NVS not initialized
[00:00:00.012,000] <err> fs_nvs: NVS not initialized
[00:00:00.012,000] <err> fs_nvs: NVS not initialized
[00:00:00.016,000] <err> fs_nvs: NVS not initialized
[00:00:00.016,000] <err> fs_nvs: NVS not initialized
[00:00:00.018,000] <err> fs_nvs: NVS not initialized
[00:00:00.018,000] <err> fs_nvs: NVS not initialized
[00:00:00.022,000] <err> fs_nvs: NVS not initialized
Reboot counter history: 
 Error, no Reboot counter
Rebooting in ...5...4...3...2...1�*** Booting Zephyr OS build zephyr-v2.2.0  ***
```

Laczen, 2020-04-14:
@akofoed, thanks for the tests. The default dts is to be expected, the sectorsize is too large and will not fit 3 sectors.

So to summarize: 
a. The dts definition for the stm32f429zi is not suited for nvs.
b. The current master fails the nvs test even if the dts definition is correct.

Could you file a bug report for item b?

For item a. could you create a PR ?

Thanks

carlescufi, 2020-04-21:
Closing since this has been covered by #24357 and the DTS for stm32f429zi will be a PR

closed 2020-04-21 by carlescufi