zephyrproject-rtos/zephyr #24243: MCUBoot not working on disco_l475_iot1

nandojve, 2020-04-09:
**Describe the bug**
```
*** Booting Zephyr OS build zephyr-v2.2.0-1280-g96f76968ebc6  ***
[00:00:00.005,000] <inf> mcuboot: Starting bootloader
[00:00:00.006,000] <wrn> mcuboot: Failed reading sectors; BOOT_MAX_IMG_SECTORS=128 - too small?
[00:00:00.006,000] <err> mcuboot: Image in the primary slot is not valid!
[00:00:00.006,000] <err> mcuboot: Unable to find bootable image
```

**To Reproduce**
Build current mainline MCUBoot/Zephyr for **disco_l475_iot1** board

**Expected behavior**
After apply this patch it worked fine.
```
diff --git a/boot/zephyr/boards/disco_l475_iot1.conf b/boot/zephyr/boards/disco_l475_iot1.conf
new file mode 100644
index 0000000..6cedc26
--- /dev/null
+++ b/boot/zephyr/boards/disco_l475_iot1.conf
@@ -0,0 +1,2 @@
+# MCUBoot settings
+CONFIG_BOOT_MAX_IMG_SECTORS=256
```
Shell:
```
*** Booting Zephyr OS build zephyr-v2.2.0-1280-g96f76968ebc6  ***
[00:00:00.005,000] <inf> mcuboot: Starting bootloader
[00:00:00.007,000] <inf> mcuboot: Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.007,000] <inf> mcuboot: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.007,000] <inf> mcuboot: Boot source: primary slot
[00:00:00.015,000] <inf> mcuboot: Swap type: none

uart:~$ *** Booting Zephyr OS build zephyr-v2.2.0-1260-g6e899acce9bb  ***
[00:00:00.539,000] <inf> net_config: Initializing network
[00:00:00.539,000] <inf> net_config: IPv4 address: 192.168.50.153
uart:~$ 

```
**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.11.2
 - mainline

**Additional context**
I not sure if the fix should be sent to MCUBoot module or it can be addressed on Zephyr side like on board, SoC or flash driver.

erwango, 2020-04-09:
^^@ABOSTM

ABOSTM, 2020-04-10:
Hi @nandojve ,
I could not find the file your are talking about 

> /boot/zephyr/boards/disco_l475_iot1.conf

https://github.com/zephyrproject-rtos/mcuboot/tree/master/boot/zephyr/boards
not even in MCU ooficial github
https://github.com/JuulLabs-OSS/mcuboot/tree/master/boot/zephyr/boards
So where does it come from ?

ABOSTM, 2020-04-10:
Hi @nandojve ,
Ok I get it. you propose a new file.
So can you create a PullRequest in zephyr, but in https://github.com/zephyrproject-rtos/mcuboot/ 

So this issue should be moved to https://github.com/zephyrproject-rtos/mcuboot/ (instead of https://github.com/zephyrproject-rtos/zephyr)

utzig, 2020-04-11:
> Hi @nandojve ,
> Ok I get it. you propose a new file.
> So can you create a PullRequest in zephyr, but in https://github.com/zephyrproject-rtos/mcuboot/

The fix is correct but you have to submit the PR here: https://github.com/JuulLabs-OSS/mcuboot

carlescufi, 2020-04-14:
@nandojve will you send a PR to zephyr's MCUboot to update our fork?

carlescufi, 2020-04-14:
Not considered a Zephyr bug since this is board support in mcuboot and has been done upstream.

PR #24387 (2020-04-17): west.yml: Sync MCUboot for disco_l475_iot1
MCUBoot not working on disco_l475_iot1.

zephyrproject-rtos/mcuboot#13
Fixes #24243.
commit a280d20b (2020-04-17): west.yml: Sync MCUboot for disco_l475_iot1

MCUBoot not working on disco_l475_iot1.

Fixes #24243.
closed 2020-04-14 by carlescufi