zephyrproject-rtos/zephyr #24255: k_cpu_idle() is called with interrupts locked inside idle()

dcpleung, 2020-04-10:
Inside `kernel/idle.c`, function `sys_power_save_idle()`, `k_cpu_idle()` is called when `_sys_suspend()` determines that it is not desirable to suspend. And `sys_power_save_idle()` is called within `idle()` in the same file when `arch_irq_lock()` is called just before. This causes issues, for example on ARM, where `k_cpu_idle()`->`arch_cpu_idle()` would manipulate `BASEPRI` where `arch_irq_lock()` also manipulate `BASEPRI`. Compounding the problem is that `idle()` never calls `arch_irq_unlock()` so `BASEPRI` is never restored. With `arch_cpu_idle()` inherits the `BASEPRI` from `arch_irq_lock()` and `arch_cpu_idle() restores the inherited `BASEPRI`, this may prevent handling of normal interrupts (as these are masked).

carlescufi, 2020-05-21:
@wentongwu any chance to get a fix for this in time for 2.3.0?

ioannisg, 2020-07-21:
> Compounding the problem is that idle() never calls arch_irq_unlock() so BASEPRI is never restored.

@dcpleung BASEPRI is unconditionally restored inside arch_cpu_idle(). 

> This causes issues, for example on ARM, where k_cpu_idle()->arch_cpu_idle() would manipulate BASEPRI where arch_irq_lock() also manipulate BASEPRI.

It is not clear what issues you refer to here.

Honestly, it is a bit hard for me to understand what the problem is for ARM architecture, I'd hope for some more clarifications here.

dcpleung, 2020-07-21:
This has been a while... if I remember correct, BASEPRI was restored incorrectly when PM was enabled. `arch_irq_lock()` writes a new value to BASEPRI in `idle()`, and `arch_cpu_idle()` is restoring this BASEPRI instead of the value before calling `arch_irq_lock()` (inside `idle()`). This results in incorrect mask.

ioannisg, 2020-07-22:
So, as far as I can see in the code, idle() does the following:
- locks interrupts without preserving the key (i.e. BASEPRI value in Cortex-M). Effectively, this means that the unlocking operation "should not care" about the actual key value and is expected to unlock all interrupts
- then calls k_cpu_idle() -> arch_cpu_idle()

@dcpleung appreciate coming back to this.

In Cortex-M arch_cpu_idle() unconditionally unlocks interrupts, setting BASEPRI to 0, so it fulfills the expectation mentioned above.

> arch_irq_lock() writes a new value to BASEPRI in idle(), and arch_cpu_idle() is restoring this BASEPRI instead of the value before calling arch_irq_lock() (inside idle()). 

This is true. But is there a problem with that? Is there a reason for the idle thread to "partially" mask IRQs? I mean, is it not expected that idle() would always be called with all interrupts unmasked? If not, then, yes, we should probably need to preserve the key when locking IRQs in idle() and restore _that_ value in arch_cpu_idle() instead of unconditional unlock. But we need to answer this question first.

> This results in incorrect mask.

This is correct, but it is not what you describe in the original comment:

>> this may prevent handling of normal interrupts (as these are masked).

What happens is that interrupts are unconditionally unlocked that is, even if they were partially masked before calling idle()

dcpleung, 2020-07-22:
Actually, I remember now that I had to workaround the incorrect mask in commit b935903543b660af1a79e551dd62fca3a87d5123. The issue only happens if deep sleep is enabled. Here is the commit message:

```
z_power_soc_deep_sleep() is called with interrupt locked already
so restoring BASEPRI is pointless here, as it would only allow
exceptions afterwards. The situation is complicated by the fact
that kernel/idle.c:idle() only locks interrupt without unlocking
which means the BASEBRI at entry of z_power_soc_deep_sleep() is
already set to allow exceptions only but not lower priority
interrupts like timer. So when, e.g. timer, interrupt fires,
the SoC would come out of deep sleep but the waking interrupts
are never delivered since they are masked, and idle() will try
to sleep again. And now it gets into a loop of going into deep
sleep briefly and waking up immediately and it goes on and on.
The solution is not to restore BASEPRI and simply leave it at
zero. This is a workaround as a proper fix would involve
invasion changes to the PM subsystem.
    
Also, _sys_pm_power_state_exit_post_ops() is not being called
when deep sleep is involved, so PRIMASK needs to be reset
after coming out of deep sleep.
```

This issue was observed on the `mec1501` SoC.

According to `arch_cpu_idle()`, setting BASEPRI to 0 seems to be the correct procedure. Maybe BASEPRI needs to set to 0 after coming out of deep sleep? But we lose the mask already being set before entering idle (... not sure if masking interrupts via BASEPRI is a reasonable thing to do...?).

dcpleung, 2020-07-22:
Would there be a use case where mask is set outside `idle()` which needs to be preserved? If not, this can probably be closed. It's that BASEPRI is not preserved across `idle()` which raises my original question.

ioannisg, 2020-07-22:
> Would there be a use case where mask is set outside `idle()` which needs to be preserved? If not, this can probably be closed. It's that BASEPRI is not preserved across `idle()` which raises my original question.

I don't believe so. Why would the Idle thread mask interrupts? Also, we don't really exit idle(), right? There's a while loop in the current implementation. I'd like to hear @andyross opinion on this. 

One exception would be if k_cpu_idle() is allowed to be called by user application; but I think, in that case it's sufficient to document that k_cpu_idle should be unconditionally unlocking IRQs upon return (which we already do).

andyross, 2020-07-23:
This flares up periodically.  Our idle API isn't very clean.  Basically the structure is (if I haven't forgotten something important):

* The top of the idle thread main loop has interrupts enabled

* An interrupt lock is taken in the idle thread as it calls the platform layer (be it the PM API or k_cpu_idle()).  This is required, as the decision to suspend needs to be atomic.  Otherwise we could take an interrupt and wake up a thread from inside the platform layer performing the suspend (which doesn't know anything about the scheduler) and the system would then enter an indefinite idle instead of running the thread unblocked by the I/O it just serviced.

* The platform layer needs to do some level of magic to make sure that interrupts are unmasked as it enters the hardware idle state, so that it can wake up.  This bit isn't the kernel's problem, really.

* Then, for reasons that are lost to history, the platform layer returns to the caller with **interrupts unmasked**.  That is, the idle thread takes an interrupt lock but never releases it.  That's... crazy.  But it works, I guess.

andyross, 2020-07-23:
Oh, but I think I see @dcpleung 's point above: the PM layer's idle call happens in two different contexts, and one of them isn't tolerant of that "unconditionally unmask interrupts" behavior that the idle loop wants.  Yeah, that sounds like a bug.

ioannisg, 2020-07-24:
> Oh, but I think I see @dcpleung 's point above: the PM layer's idle call happens in two different contexts, and one of them isn't tolerant of that "unconditionally unmask interrupts" behavior that the idle loop wants. Yeah, that sounds like a bug.

@andyross can you help me understand why the PM's idle call is not tolerant?

To me, there seems to be single call to _sys_suspend() from PM API (the other call is from kernel/idle.c). Here is what I see:

```
void sys_pm_force_power_state(enum power_states state)
{
	__ASSERT(state >= SYS_POWER_STATE_AUTO &&
		 state <  SYS_POWER_STATE_MAX,
		 "Invalid power state %d!", state);

#ifdef CONFIG_SYS_PM_DIRECT_FORCE_MODE
	(void)arch_irq_lock();
	forced_pm_state = state;
	_sys_suspend(K_TICKS_FOREVER);
#else
	forced_pm_state = state;
#endif
}
```

So the IRQ's are locked, without key-preservation before _sys_suspend() is called, and _sys_suspend() will eventually call z_power_soc_sleep() or _deep_sleep() which mimics what k_cpu_idle() does. So both k_cpu_idle() and SoC PM sleep functions unconditionally unlock interrupts upon return. And their callers expect that, since they are not releasing the lock. So unless I miss something, there does not seem to be any inconsistency here.

To clarify @andyross @dcpleung , I am not at all against refining the Idle API, just that I cannot associate this ticket's description with a bug (other than an un-clean API)

> Would there be a use case where mask is set outside idle() which needs to be preserved? If not, this can probably be closed. It's that BASEPRI is not preserved across idle() which raises my original question.

This is the million dollar question, IMHO. In theory, you might want to set an interrupt mask before entering idle(), so you limit the external sources, by which the CPU is allowed to wake up. In that case, the IRQ mask needs to be preserved after exiting _sys_suspend(). But I have not found a use-case for this hypothetical scenario.

I'd say let's make this issue an enhancement instead, to look deeper into the requirements of the idle API.

andyross, 2020-07-24:
I could get behind that too.  You're right, that code looks correct as long as we're reasoning correctly about the calling contexts.  At worst it's fragile.

FWIW: if we get around to fixing this in all the idle/suspend implemenations, the Right Thing here would be for the idle call to take the irq lock key and release it on behalf of the caller, the way that z_swap() (and all its rescheduling wrappers) does.  That way it acts as a condition variable and no one gets confused about the locking state.

andyross, 2020-07-24:
(Actually probably best to make it a spinlock while we do it, because inevitably some SMP platform is going to have suspend state bookeeping that needs to be synchronized across all the CPUs)

dcpleung, 2020-07-24:
I am okay with it not being a bug. On MEC1501, deep sleep is currently working fine now without any reported issues.

ioannisg, 2020-07-25:
> I am okay with it not being a bug. On MEC1501, deep sleep is currently working fine now without any reported issues.

Will do this, @dcpleung . I will re-describe this ticket, so it focuses on cleaning the API.

sandeepbrcm, 2020-07-31:
This discussion seems relevant to a confusion I have at the moment.
   ```
     while (true) {     /*  <<  we should probably check here if idle is not the 
#if SMP_FALLBACK
                k_busy_wait(100);
                k_yield();
#else
                (void)arch_irq_lock();   /* what if there is already one interruption before this point and ISR was serviced, when will it yield */
                sys_power_save_idle();
    
                IDLE_YIELD_IF_COOP();
                /*  when I return from k_yield() what should be the state of my interrupts .  I must expect them to be masked  
                 *  Otherwise there must be some other mechanism to flag weather an yield is needed before we attempt idle again.
                 */
#endif
        }
```
 When the idle task is context-switched in the interrupts are unmasked.  ie.  an interrupt can come and ISR may make another thread ready. However in coop mode we wont be switching out and we will continue in idle task. which will eventually enter
idle state ( eg: wfi or so) unexpectedly. 

Am I missing something here.  This is probably applicable for all arch.  If the understanding is correct this PR might help ref:  #27297

wentongwu, 2020-09-28:
per @jocelyn-li's request, update the status here: I haven’t got any time to review the comments, make the effort estimation and debug it so far because of other high priority issues assigned to me. 
cc @nashif @andrewboie @rljordan

Icefirejm, 2024-02-21:
I'm running into the same issue as was described here earlier, wherein a wake-up timer fires an interrupt and the system wakes up and promptly tries going back to sleep from idle() and stays in a loop of waking up and going back to sleep.
Are there any solutions to this? Setting BASEPRI to 0 after wakeup doesn't work.

ceolin, 2024-04-27:
@Icefirejm apologies for the late reply. Can you tell me what is the revision, target and configuration are you using ? I tried to reproduce it but I couldn't. Even with preemption disabled. This issue was reported a long time ago and this code has changed a lot, though the general idea kept the same.